stupidloud/nanopi-openwrt #323: minimal固件中的passwall不支持DOH

EsuRt, 2020-12-03:
minimal固件中的passwall无法设置DOH，提示“自定义DNS 不是有效IP格式，请重新输入！”，但看了下xiaorouji/openwrt-passwall的配置，应该已经支持DOH了，请问下应该怎样设置？

avsba001, 2020-12-05:
minimal里面的pw是改openclash吧，想用doh可以用smartdns曲线救国

EsuRt, 2020-12-05:
add_passwall.sh 文件中用的就是xiaorouji/openwrt-passwall。
```
echo -e '\nsrc-git lienol https://github.com/xiaorouji/openwrt-passwall' >> feeds.conf.default
./scripts/feeds update lienol
rm -rf feeds/lienol/lienol/ipt2socks
rm -rf feeds/lienol/lienol/shadowsocksr-libev
rm -rf feeds/lienol/lienol/pdnsd-alt
rm -rf feeds/lienol/lienol/luci-app-verysync
rm -rf feeds/lienol/package/verysync
#rm -rf feeds/lienol/package/chinadns-ng
rm -rf package/lean/openwrt-chinadns-ng
rm -rf package/lean/luci-app-kodexplorer
rm -rf package/lean/luci-app-pppoe-relay
rm -rf package/lean/luci-app-pptp-server
rm -rf package/lean/luci-app-v2ray-server
./scripts/feeds install -a -p lienol
```
https://github.com/klever1988/nanopi-openwrt/blob/b2d851c85af763c8405e324b0e374688ba9b2d4f/add_passwall.sh

closed 2020-12-18 by EsuRt