stupidloud/nanopi-openwrt #326: 最近几天的固件，没有一个能进后台的，DHCP全都有问题，拔电源重启一个样

gimgoo, 2020-12-04:
最近几天的固件，没有一个能进后台的，DHCP全都有问题，拔电源重启一个样

avsba001, 2020-12-05:
我是12月1号的Lienol版本，重刷先用工具把tf写1G左右的0数据，刷好后开DHCP，不要开acc加速，现在很正常

Wyaohui, 2020-12-07:
> 我是12月1号的Lienol版本，重刷先用工具把tf写1G左右的0数据，刷好后开DHCP，不要开acc加速，现在很正常

原理是啥？我也是，最近固件经常进不到后台，网络端口也不识别，只能知己改ip然后碰运气进

avsba001, 2020-12-09:
> > 我是12月1号的Lienol版本，重刷先用工具把tf写1G左右的0数据，刷好后开DHCP，不要开acc加速，现在很正常
> 
> 原理是啥？我也是，最近固件经常进不到后台，网络端口也不识别，只能知己改ip然后碰运气进

[image]
为了保险我都这样操作，除了404大佬的超频固件我启动不了其他都还好

avsba001, 2020-12-09:
> > 我是12月1号的Lienol版本，重刷先用工具把tf写1G左右的0数据，刷好后开DHCP，不要开acc加速，现在很正常
> 
> 原理是啥？我也是，最近固件经常进不到后台，网络端口也不识别，只能知己改ip然后碰运气进

[image]
不是很懂，我理解的是像格式化那样，数据并没有丢，刷入同一个人的固件可能污染现有配置

closed 2020-12-10 by gimgoo