stupidloud/nanopi-openwrt #337: 2020.12.14的固件在线刷机失败！

supperups, 2020-12-14:
以前刷机一直都是成功的，但是今天刷机两次都是失败。请看以下信息：

准备重新打包
Error loading shared library libzstd.so.1: No such file or directory (needed by /usr/bin/zstdmt)
Error relocating /usr/bin/zstdmt: ZSTD_createDStream: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_initDStream_usingDict: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_DCtx_setMaxWindowSize: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_compress2: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_DStreamOutSize: symbol not found
Error relocating /usr/bin/zstdmt: ZDICT_trainFromBuffer_fastCover: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_minCLevel: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_CCtx_setParameter: symbol not found
Error relocating /usr/bin/zstdmt: ZDICT_optimizeTrainFromBuffer_cover: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_resetDStream: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_isError: symbol not found
Error relocating /usr/bin/zstdmt: ZDICT_trainFromBuffer_unsafe_legacy: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_frameHeaderSize: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_getErrorName: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_sizeof_CCtx: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_freeCStream: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_freeDStream: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_XXH64: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_createDCtx: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_getErrorCode: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_freeDCtx: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_cycleLog: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_findDecompressedSize: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_CCtx_setPledgedSrcSize: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_isFrame: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_getCParams: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_DCtx_reset: symbol not found
Error relocating /usr/bin/zstdmt: ZDICT_optimizeTrainFromBuffer_fastCover: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_decompressStream: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_compressBound: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_getFrameProgression: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_compressStream2: symbol not found
Error relocating /usr/bin/zstdmt: ZDICT_getErrorName: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_cParam_getBounds: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_maxCLevel: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_CStreamInSize: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_freeCCtx: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_toFlushNow: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_CStreamOutSize: symbol not found
Error relocating /usr/bin/zstdmt: ZDICT_trainFromBuffer_cover: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_getFrameContentSize: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_CCtx_refPrefix: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_CCtx_loadDictionary: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_createCCtx: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_CCtx_reset: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_DCtx_loadDictionary: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_getFrameHeader: symbol not found
Error relocating /usr/bin/zstdmt: ZDICT_isError: symbol not found
Error relocating /usr/bin/zstdmt: ZSTD_DStreamInSize: symbol not found
打包完毕，准备刷机
root@FriendlyWrt:/#

supperups, 2020-12-14:
根本没有刷机

TypeVII, 2020-12-15:
libzstd 有了更新
opkg install libzstd