stupidloud/nanopi-openwrt #349: 这几天R2固件编译

w261643358, 2020-12-21:


w261643358, 2020-12-21:
这几天R2S固件编译各种报错

1263478456, 2020-12-22:
是用的那个minimal的自动升级脚本吗？我之前发现那个自动升级脚本有一个问题，就是只能更新最近一天出的更新，但是最近minimal已经有好几天没更新了，所以自动更新脚本用不了

MaxwellXiao, 2020-12-22:
> 是用的那个minimal的自动升级脚本吗？我之前发现那个自动升级脚本有一个问题，就是只能更新最近一天出的更新，但是最近minimal已经有好几天没更新了，所以自动更新脚本用不了

是的，好像只读了前一天的固件，没加循环

supperups, 2020-12-23:
已经连续多天没有更新了

w261643358, 2020-12-23:
GITHUB自动编译的跑一分多钟就提示报错了  貌似有源码更新 这边的脚本没有及时同步更新 咱又不懂编程只能傻傻的等

kid424, 2020-12-23:
踩了一下坑，作者要修复的话可以节省点时间：
1. 官方的install.sh开头加了一句不是安装包的语句，导致1_initialization_environment.sh里面执行install.sh的时候报错。可以在sed里加上删除这行或者直接跑那个sh
2. patch_kernel_5.4.sh里要用到的https://github.com/armbian/build/raw/master/patch/kernel/rockchip64-dev/RK3328-enable-1512mhz-opp.patch被删了，可以用https://github.com/FireflyTeam/armbian-build/raw/master/patch/kernel/rockchip64-dev/RK3328-enable-1512mhz-opp.patch替代
3. lean稳定版固件里autocore的文件路径进行了更改，分出了x86和arm两个文件夹，导致enable_autocore.diff补丁应用失败。不太确定这个除了路径需不需要改别的

JorWang, 2020-12-23:
我看这个作者弄的没问题 https://github.com/biliwala/nanopi-openwrt

w261643358, 2020-12-24:
开始干活了各位!