stupidloud/nanopi-openwrt #356: NAT 回环问题

woOzZ1, 2020-12-27:


woOzZ1, 2020-12-27:
问题得到了解决，原因是有错误的提交开启了bridge防火墙，导致回流的流量被阻止。

woOzZ1, 2020-12-27:
https://github.com/coolsnowwolf/lede/issues/2103

d0zingcat, 2021-08-11:
> 问题得到了解决，原因是有错误的提交开启了bridge防火墙，导致回流的流量被阻止。

你好，可以详细说说这个怎么解决么？我也碰到了同样的问题，各种措施都试过了但是还是无法正常回环

woOzZ2, 2021-08-12:
> > 问题得到了解决，原因是有错误的提交开启了bridge防火墙，导致回流的流量被阻止。
> 
> 你好，可以详细说说这个怎么解决么？我也碰到了同样的问题，各种措施都试过了但是还是无法正常回环

原因不是`NAT loopback`不起作用，而是bridge防火墙给阻挡了回流的流量，之前一直怀疑是这样，因为发现nmap一直显示`filtered`而不是`closed`
如已经编译可以在`/etc/sysctl.conf `和`/etc/sysctl.d/11-br-netfilter.conf` 修改
```
net.bridge.bridge-nf-call-arptables=0
net.bridge.bridge-nf-call-ip6tables=0
net.bridge.bridge-nf-call-iptables=0
```
然后`sysctl -p` 即可使用NAT loopback

链接都贴出来了你点进去看看就有了啊

d0zingcat, 2021-08-12:
> 原因不是`NAT loopback`不起作用，而是bridge防火墙给阻挡了回流的流量，之前一直怀疑是这样，因为发现nmap一直显示`filtered`而不是`closed`
> 如已经编译可以在`/etc/sysctl.conf `和`/etc/sysctl.d/11-br-netfilter.conf` 修改
>
> ```
> net.bridge.bridge-nf-call-arptables=0 [...]

嗷嗷，这个我已经看过了（我以为你说的解决方法不单单是这个）。我的机器本来就都是0，但还是存在问题。而且诡异的是，只要开tcpdump就是通的。

d0zingcat, 2021-08-13:
> 原因不是`NAT loopback`不起作用，而是bridge防火墙给阻挡了回流的流量，之前一直怀疑是这样，因为发现nmap一直显示`filtered`而不是`closed`
> 如已经编译可以在`/etc/sysctl.conf `和`/etc/sysctl.d/11-br-netfilter.conf` 修改
>
> ```
> net.bridge.bridge-nf-call-arptables=0 [...]

sorry我查资料的时候先查到的你贴的那个issue并且当时试过了不行然后找到的这个贴所以以为你的解决办法还有别的。
又检查了下发现`/etc/sysctl.conf`这个文件里面只有两行，少了`net.bridge.bridge-nf-call-arptables=0`这个。我手动加上了之后果然可以了，感谢🙏

godotu, 2022-04-28:
终于找到解决的方案了，感谢

socekin, 2022-05-03:
> 原因不是`NAT loopback`不起作用，而是bridge防火墙给阻挡了回流的流量，之前一直怀疑是这样，因为发现nmap一直显示`filtered`而不是`closed` 如已经编译可以在`/etc/sysctl.conf `和`/etc/sysctl.d/11-br-netfilter.conf` 修改
>
> ```
> net.bridge.bridge-nf-call-arptables=0 [...]

按这个方法编辑后，报错

sysctl: error: 'net.bridge/bridge-nf-call-arptables' is an unknown key

uclee888, 2022-05-11:
解决了我的问题，感谢

steamch, 2022-08-15:
> 按这个方法编辑后，报错
>
> sysctl: error: 'net.bridge/bridge-nf-call-arptables' is an unknown key

我也遇到了这个报错

woOzZ2, 2024-03-10:
> 按这个方法编辑后，报错
>
> sysctl: error: 'net.bridge/bridge-nf-call-arptables' is an unknown key

可能是内核缺少网桥模块
试试
```
lsmod | grep bridge
modprobe bridge
lsmod | grep bridge
```

如果第一个lsmod没有输出而第三条命令的lsmod有输出应该可以添加了

woOzZ2, 2024-03-10:
> 我也遇到了这个报错

可能是内核缺少网桥模块
试试
```
lsmod | grep bridge
modprobe bridge
lsmod | grep bridge
```

如果第一个lsmod没有输出而第三条命令的lsmod有输出应该可以添加了

steamch, 2024-03-10:
> 可能是内核缺少网桥模块 试试
>
> ```
> lsmod | grep bridge
> modprobe bridge
> lsmod | grep bridge
> ```
>
> 如果第一个lsmod没有输出而第三条命令的lsmod有输出应该可以添加了

`[root@OpenWrt:02:27 AM ~] # modprobe bridge
failed to find a module named bridge`
提示我这个
有一个桥接配置
`[root@OpenWrt:02:32 AM ~] # brctl show
bridge name     bridge id               STP enabled     interfaces
br-lan          7fff.f6ec1554a502       no              eth0
`

linchangyi, 2025-03-13:
> 可能是内核缺少网桥模块 试试
>
> ```
> lsmod | grep bridge
> modprobe bridge
> lsmod | grep bridge
> ```
>
> 如果第一个lsmod没有输出而第三条命令的lsmod有输出应该可以添加了

我也遇到了这个问题，执行 `sysctl -p` 之后报错
```
net.bridge.bridge-nf-call-arptables = 0
net.bridge.bridge-nf-call-ip6tables = 0
net.bridge.bridge-nf-call-iptables = 0
```
执行 lsmod | grep bridge  有输出
执行 modprobe bridge 提示 failed to find a module named bridge

现在有啥解决办法吗