stupidloud/nanopi-openwrt #365: 这个固件是不是以后不更新了啊，大家还有没在更新的固件啊？

21967201, 2021-01-23:
这个固件是不是以后不更新了啊，大家还有没在更新的固件啊？

VKF47, 2021-01-24:
好像openwrt官方直接支持R2S了，可以去折腾一下

VKF47, 2021-01-25:
找到一个不错的，https://github.com/QiuSimons/R2S-R4S-X86-OpenWrt

oldoldstone, 2021-01-25:
这个可以，集成了docker
https://github.com/biliwala/nanopi-openwrt，

我自己也fork了一个，
https://github.com/oldoldstone/nanopi-openwrt
按自己的习惯，我主要加了
nfs
transmission
ntfs-3g
vsftp
docker

blizzardster, 2021-01-28:
> 找到一个不错的，https://github.com/QiuSimons/R2S-R4S-X86-OpenWrt

谢谢，我下载了试一下，

zdante, 2021-02-06:
> 找到一个不错的，https://github.com/QiuSimons/R2S-R4S-X86-OpenWrt

这个我用过，这个固件操作保存什么的都太慢了，并且不支持无线网卡。当然，也可能是没有支持我的无线网卡。。。固件我几乎刷遍，最好用的还是这个klever1988大佬的固件。稳定，操作上也比较顺畅，我的无线网卡也能顺利驱动，但目前我的网卡无法使用5g功能，可惜这个大佬不更新了。。。

guicomeon, 2021-02-07:
> 固件我几乎刷遍，最好用的还是这个klever1988大佬的固件。稳定，操作上也比较顺畅
大佬目前在用的版本是哪个？我懒得折腾，家里还在用2020-10-16的，运行几天感觉偶尔不稳定，想升级一下。

zdante, 2021-02-07:
> > 固件我几乎刷遍，最好用的还是这个klever1988大佬的固件。稳定，操作上也比较顺畅
> 
> 大佬目前在用的版本是哪个？我懒得折腾，家里还在用2020-10-16的，运行几天感觉偶尔不稳定，想升级一下。

我目前用的就是这个大佬的最新一版，感觉很稳定。唯一遗憾就是我的无线网卡无法驱动5g功能。。。很可惜klever1988大佬目前断更了。。。

TypeVII, 2021-02-08:
接近2个月无重启无死机，主要是稳定而不是天天更新。

zhixiangjoy, 2021-02-08:
> 找到一个不错的，https://github.com/QiuSimons/R2S-R4S-X86-OpenWrt

r2s刷了最新两天的固件无法启动，刷三天前的版本，风扇一直转，最后放弃了

zdante, 2021-02-08:
> > 找到一个不错的，https://github.com/QiuSimons/R2S-R4S-X86-OpenWrt
> 
> r2s刷了最新两天的固件无法启动，刷三天前的版本，风扇一直转，最后放弃了

404的固件他的wan口lan口是颠倒的，你要是插错了也没显示，他的那个固件真的好卡。。。

VKF47, 2021-02-08:
> > > 找到一个不错的，https://github.com/QiuSimons/R2S-R4S-X86-OpenWrt
> > 
> > 
> > r2s刷了最新两天的固件无法启动，刷三天前的版本，风扇一直转，最后放弃了
> 
> 404的固件他的wan口lan口是颠倒的，你要是插错了也没显示，他的那个固件真的好卡。。。

好像是颠倒更好，毕竟有一个是USB转，卡的话还可以，因为只要设置没错等个5秒吧，设置错了90秒后提示回滚。

VwEl, 2021-02-15:
> 
> 
> 这个可以，集成了docker
> [https://github.com/biliwala/nanopi-openwrt，](https://github.com/biliwala/nanopi-openwrt%EF%BC%8C)
> 
> 我自己也fork了一个，
> https://github.com/oldoldstone/nanopi-openwrt
> 按自己的习惯，我主要加了
> nfs
> transmission
> ntfs-3g
> vsftp
> docker

问一下1G内存跑docker够用嘛

oldoldstone, 2021-02-17:
> > 这个可以，集成了docker
> > [https://github.com/biliwala/nanopi-openwrt，](https://github.com/biliwala/nanopi-openwrt%EF%BC%8C)
> > 我自己也fork了一个，
> > https://github.com/oldoldstone/nanopi-openwrt
> > 按自己的习惯，我主要加了
> > nfs
> > transmission
> > ntfs-3g
> > vsftp
> > docker
> 
> 问一下1G内存跑docker够用嘛

看你跑啥了，我就跑一个nextcloud+mysql，文件不多，没啥问题，内存才消耗300多M

zhixiangjoy, 2021-02-18:
> > > 这个可以，集成了docker
> > > [https://github.com/biliwala/nanopi-openwrt，](https://github.com/biliwala/nanopi-openwrt%EF%BC%8C)
> > > 我自己也fork了一个，
> > > https://github.com/oldoldstone/nanopi-openwrt
> > > 按自己的习惯，我主要加了
> > > nfs
> > > transmission
> > > ntfs-3g
> > > vsftp
> > > docker
> > 
> > 
> > 问一下1G内存跑docker够用嘛
> 
> 看你跑啥了，我就跑一个nextcloud+mysql，文件不多，没啥问题，内存才消耗300多M

你们打开防火墙页面卡吗？ 我打开防火墙页面不能马上出来，要卡一会，不知道是不是内存卡的原因？

lhf19900402, 2021-03-01:
> > 找到一个不错的，https://github.com/QiuSimons/R2S-R4S-X86-OpenWrt
> 
> r2s刷了最新两天的固件无法启动，刷三天前的版本，风扇一直转，最后放弃了

你后来解决了吗？

bg4drl, 2021-03-01:
klever1988不是刚刚在改 Auto ARM core  快了 
R4S已经可用

zhixiangjoy, 2021-03-01:
> > > 找到一个不错的，https://github.com/QiuSimons/R2S-R4S-X86-OpenWrt
> > 
> > 
> > r2s刷了最新两天的固件无法启动，刷三天前的版本，风扇一直转，最后放弃了
> 
> 你后来解决了吗？

没有，放弃404大佬的固件了，用的其他人的固件。