stupidloud/nanopi-openwrt #371: What's app call doesn't work

xenstar, 2021-02-17:
What's app call is always stuck in "connecting". My UDP server is turned on. I have attached the setting.
[image]

Anyways to fix it?

xenstar, 2021-02-18:
There is some issue with UDP. I check with this website https://networktest.twilio.com/ also failed. 

Please check. In SSR+, I have already enabled UDP (Selected Same as Server)

[image]

**Here is my setting** 

[image]

klever1988, 2021-02-27:
You can try to upgrade the geographic IP database from this APP, or switch to other route mode.

xenstar, 2021-02-28:
"from this APP" this app what do you mean? You can reply to me in Chinese.

klever1988, 2021-02-28:
ssrp插件有ip数据库不是，升级一下看看，如果不行切换ssrp的运行模式到全局试试看是不是udp不支持

xenstar, 2021-03-01:
Hi, Thanks for the quick reply. Well I have updated the IP Database, also tried with Global Mode, doesn't work. Tried with telegram call, what's app call, it's stuck at "connecting". You can have a try actually. 

When I am using same server, on phone with v2rayNG, it works. So I am using there is nothing wrong with the server, but issue with some settings on the firmware.

xenstar, 2021-03-03:
Change to another firmware, and it's working. The issue is still with the firmware, hopefully you can check it sometimes and fix it. All the best.

closed 2021-03-03 by xenstar