stupidloud/nanopi-openwrt #388: 关于为何接下级无线路由后无法科学上网。。谢谢

maiapig, 2021-03-04:
你好，我将R2S的WAN连接光猫，LAN连接下一级无线路由器的WAN口，在开启SSRplus后可以通过R2S后台的谷歌检查连通性。

讲道理这时候接入R2S下无线路由器的设备应该可以翻Q了，但是手机在连接无线路由器后无法科学上网，请问是否还什么地方需要特别设置吗？谢谢

ymai6, 2021-03-05:
我也一样，我用的passwall，用的vess，更新4号和5号的固件科学上网插件也连不上

klever1988, 2021-03-05:
> 我也一样，我用的passwall，用的vess，更新4号和5号的固件科学上网插件也连不上

今天发布的旧版本xray有问题，已经替换掉了。重新刷吧

klever1988, 2021-03-05:
> 你好，我将R2S的WAN连接光猫，LAN连接下一级无线路由器的WAN口，在开启SSRplus后可以通过R2S后台的谷歌检查连通性。
> 
> 讲道理这时候接入R2S下无线路由器的设备应该可以翻Q了，但是手机在连接无线路由器后无法科学上网，请问是否还什么地方需要特别设置吗？谢谢

hmm 你的无线路由可能内置了dns缓存的功能 尝试把手机的wifi dns设置为r2s的地址试试

maiapig, 2021-03-05:
> > 你好，我将R2S的WAN连接光猫，LAN连接下一级无线路由器的WAN口，在开启SSRplus后可以通过R2S后台的谷歌检查连通性。
> > 讲道理这时候接入R2S下无线路由器的设备应该可以翻Q了，但是手机在连接无线路由器后无法科学上网，请问是否还什么地方需要特别设置吗？谢谢
> 
> hmm 你的无线路由可能内置了dns缓存的功能 尝试把手机的wifi dns设置为r2s的地址试试

我尝试了更改手机DNS以及更改R2S下路由器的DNS，都不能成功。。

klever1988, 2021-03-06:
> > > 你好，我将R2S的WAN连接光猫，LAN连接下一级无线路由器的WAN口，在开启SSRplus后可以通过R2S后台的谷歌检查连通性。
> > > 讲道理这时候接入R2S下无线路由器的设备应该可以翻Q了，但是手机在连接无线路由器后无法科学上网，请问是否还什么地方需要特别设置吗？谢谢
> > 
> > 
> > hmm 你的无线路由可能内置了dns缓存的功能 尝试把手机的wifi dns设置为r2s的地址试试
> 
> 我尝试了更改手机DNS以及更改R2S下路由器的DNS，都不能成功。。

有没有可能是你用了国产浏览器 太和谐了

maiapig, 2021-03-06:
> > > > 你好，我将R2S的WAN连接光猫，LAN连接下一级无线路由器的WAN口，在开启SSRplus后可以通过R2S后台的谷歌检查连通性。
> > > > 讲道理这时候接入R2S下无线路由器的设备应该可以翻Q了，但是手机在连接无线路由器后无法科学上网，请问是否还什么地方需要特别设置吗？谢谢
> > > 
> > > 
> > > hmm 你的无线路由可能内置了dns缓存的功能 尝试把手机的wifi dns设置为r2s的地址试试
> > 
> > 
> > 我尝试了更改手机DNS以及更改R2S下路由器的DNS，都不能成功。。
> 
> 有没有可能是你用了国产浏览器 太和谐了

呃。。safari和pc端都试过了。。

klever1988, 2021-03-07:
关闭无线路由器的dhcp，和r2s互联lan口，也就是把无线路由当作交换机试，试过没

maiapig, 2021-03-09:
> 关闭无线路由器的dhcp，和r2s互联lan口，也就是把无线路由当作交换机试，试过没

emmm确实可以了 谢谢大佬 不过不明白为什么会有这个问题呢？另外想问一下，自动发布的版本如何进行升级呢，我现在用的是mininal版本。。是使用opkg命令更新吗？

klever1988, 2021-03-12:
https://github.com/klever1988/nanopi-openwrt#%E7%BB%88%E7%AB%AF%E5%86%85%E5%9C%A8%E7%BA%BF%E5%8D%87%E7%BA%A7%E6%96%B9%E6%B3%95

这里说了

closed 2021-03-21 by maiapig