stupidloud/nanopi-openwrt #394: 大神,新版本的固件是不是有问题呀？

ICHIGO9, 2021-03-05:
mount: /mnt/img: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.

klever1988, 2021-03-05:
你从哪个版本尝试升级的 我去试试

NanJishen, 2021-03-06:
不只是无法在线升级，我下载 r2s.img.gz 用7z也解压不了，解压到最后失败，主要是无法解压吧

klever1988, 2021-03-06:
因为昨天尝试扩大了分区容量导致解压后的镜像体积特别大，无法在tmp分区上完成升级。

已经恢复到以前的分区大小了。

WilsonXu0512, 2021-03-06:
> 因为昨天尝试扩大了分区容量导致解压后的镜像体积特别大，无法在tmp分区上完成升级。
> 
> 已经恢复到以前的分区大小了。

大神你好，分区是不是留的太小了，今天用最新版卡刷完之后，软件包剩余空间已经没有了，什么操作都没法做

klever1988, 2021-03-07:
> > 因为昨天尝试扩大了分区容量导致解压后的镜像体积特别大，无法在tmp分区上完成升级。
> > 已经恢复到以前的分区大小了。
> 
> 大神你好，分区是不是留的太小了，今天用最新版卡刷完之后，软件包剩余空间已经没有了，什么操作都没法做

这个问题很难弄，分区留大了img解压出来体积也大，刷卡写入也增加，还有就是autoupdate.sh脚本也无法使用。 留小了，目前除了把tf卡剩余空间多分一个区，没有别的办法

WilsonXu0512, 2021-03-07:
> > > 因为昨天尝试扩大了分区容量导致解压后的镜像体积特别大，无法在tmp分区上完成升级。
> > > 已经恢复到以前的分区大小了。
> > 
> > 
> > 大神你好，分区是不是留的太小了，今天用最新版卡刷完之后，软件包剩余空间已经没有了，什么操作都没法做
> 
> 这个问题很难弄，分区留大了img解压出来体积也大，刷卡写入也增加，还有就是autoupdate.sh脚本也无法使用。 留小了，目前除了把tf卡剩余空间多分一个区，没有别的办法

我已经刷回去年年底的固件了，刷回去发现去年的固件好像是自动扩容的（sd卡8g，刷完软件包那里剩7.1g左右）。最近几版固件刷完，软件包那里只剩几十M

klever1988, 2021-03-07:
嗯之前友善的内核是自带扩容的  现在的原生op需要自己操作重新挂载overlay比较麻烦

WilsonXu0512, 2021-03-07:
> 嗯之前友善的内核是自带扩容的 现在的原生op需要自己操作重新挂载overlay比较麻烦

overlay对于新手来说很不友好，而且在线升级后的处理也很麻烦，我提供一个折中的思路给您参考您看是否可行：现在img解压完大概400m左右，您把分区留到大概img包解压完后1G（甚至2G）左右，因为大多数人现在用的卡都是4g.8g或者更大，这么留的话，虽然卡容量有浪费，但是自动升级和平时在固件里折腾插件都能照顾到。

klever1988, 2021-03-07:
> overlay对于新手来说很不友好，而且在线升级后的处理也很麻烦，我提供一个折中的思路给您参考您看是否可行：现在img解压完大概400m左右，您把分区留到大概img包解压完后1G（甚至2G）左右，因为大多数人现在用的卡都是4g.8g或者更大，这么留的话，虽然卡容量有浪费，但是自动升级和平时在固件里折腾插件都能照顾到。

autoupdate.sh需要用到内存分区tmp，但是这个分区本身就有一些临时文件，加上系统内存占用，可用空间在750M以内，，留给img还能增加的容量也就350M吧。  其实如果用这个
[image]
op自带的sysupgrade升级的话，应该是不用在意img大小的，但是这个升级功能有点问题，选择“保留配置”的话升级完会有奇奇怪怪的故障。。

目前应该是没有啥办法，想扩容分区只能拔卡离线扩容了。
你一般都玩什么插件，另建分区不行吗需要在根分区上扩容？

WilsonXu0512, 2021-03-08:
> autoupdate.sh需要用到内存分区tmp，但是这个分区本身就有一些临时文件，加上系统内存占用，可用空间在750M以内，，留给img还能增加的容量也就350M吧。 其实如果用这个
> [image]
> op自带的sysupgrade升级的话，应该是不用在意img大小的，但是这个升级功能有点问题，选择“保留配置”的话升级完会有奇奇怪怪的故障。。 [...]

我是在更新adguardhome核心的时候发现容量不足这个问题的，可能是我不太会操作，我过几天再试试能不能挂载到别的位置

klever1988, 2021-03-08:
[autoupdate.sh.txt](https://github.com/klever1988/nanopi-openwrt/files/6098619/autoupdate.sh.txt)
@WilsonXu0512 
研究了一个晚上，应该可以了，我暂时没有放到公共目录，
你可以先试一下，把文件传到r2s，文件名去掉.txt，
然后执行sh autoupdate.sh应该就能扩容升级了

WilsonXu0512, 2021-03-08:
> [autoupdate.sh.txt](https://github.com/klever1988/nanopi-openwrt/files/6098619/autoupdate.sh.txt)
> @WilsonXu0512
> 研究了一个晚上，应该可以了，我暂时没有放到公共目录，
> 你可以先试一下，把文件传到r2s，文件名去掉.txt，
> 然后执行sh autoupdate.sh应该就能扩容升级了

感谢大神，扩容现在没有问题了，终于能完整测试了，但是测试出来有别的问题，我另开issue了