stupidloud/nanopi-openwrt #404: 3 月 6 号 SSRP 配 VLESS 节点出国不正常问题

HaoyiDing, 2021-03-09:
3 月 9 号用 HelloWorld 后问题消失，只是反馈一下。

## 网络环境
* 桥接改联通光猫，Lan IP 192.168.1.1 网段
* R2s 直联并拨号，Lan IP 192.168.1.1 网段
* R2s Lan -> Redmi AC 2100 WAN
* Redmi AC2100 中继模式作 AP 用
* SSRP 配置 VLESS 节点使用绕过国内 IP 模式
* 网络加速勾选 DNS 加速和缓存

## 预期
网络中所有设备出国网络全部走 VLESS 节点，成功看 Netflix 和 Youtube 8K

## 实际
Stage 1：网络中 IPv6，IPv4 都走 VLESS 节点出国，[](http://test-ipv6.com/)查看 v4、v6 地址是 VLESS 节点信息，Netflix 走 v6 隧道可看 和 Youtube 8K 流畅；
Stage 2:  网络中 IPv4 走 VLESS 节点出国，[](http://test-ipv6.com/)查看 v4 地址是 VLESS 节点信息，v6 为运营商 IP，Netflix Not Available 且 Youtube 降速明显，高清基本不可看

klever1988, 2021-03-09:
ssrp是不会处理ipv6流量的，所以Lean大佬提供了一个特性就是屏蔽v6域名的解析
https://github.com/coolsnowwolf/lede/commit/d5cd053f35928e03082121d06cf7bbea5c926b8b
[image]

这个特性在这个固件所用的immortalwrt分支也是有引入的

vssr也应该是不会处理v6的流量，他可能是用了什么办法屏蔽了v6的解析

HaoyiDing, 2021-03-09:
> ssrp是不会处理ipv6流量的，所以Lean大佬提供了一个特性就是屏蔽v6域名的解析
> [coolsnowwolf/lede@d5cd053](https://github.com/coolsnowwolf/lede/commit/d5cd053f35928e03082121d06cf7bbea5c926b8b)
> [image]
> 
> 这个特性在这个固件所用的immortalwrt分支也是有引入的
> 
> vssr也应该是不会处理v6的流量，他可能是用了什么办法屏蔽了v6的解析

谢谢答疑，基本上明白 v6 的问题所在了，我跟据我自己环境看一下重启后为什么会有一段时间 v6 是走隧道的。