stupidloud/nanopi-openwrt #406: Klever1988 老大，能否继续给出minimal的固件啊？用习惯了，这个新的感觉没有之前的稳定，经常死机。

Deckard-f-Cain, 2021-03-10:
Klever1988 老大，能否继续给出minimal的固件啊？用习惯了，这个新的感觉没有之前的稳定，经常死机。

pluter, 2021-03-10:
我刷了好几天的固件也是也经常死机-重启

klever1988, 2021-03-10:
应该是解锁1.6g频率而电压又被调低导致，我会把电压调正常，你也可以把最高频率限制一下

[image]

pluter, 2021-03-11:
> 应该是解锁1.6g频率而电压又被调低导致，我会把电压调正常，你也可以把最高频率限制一下
> 
> [image]

终于不死机啦,谢谢大佬

slb116763814, 2021-03-12:
@Deckard-f-Cain 哈哈，我也发现了，用了几次感觉新的固件都不稳定。我还是换回R2S-Minimal-2020-12-14这个固件。Minima的最后一个固件最稳定。

lyjhyjyes, 2021-04-06:
> 应该是解锁1.6g频率而电压又被调低导致，我会把电压调正常，你也可以把最高频率限制一下
> 
> [image]

老大，之前您的minimal版本很稳定，一直没升级。心血来潮我刷了r2s 2021/4/2固件，一直各种设置卡机，直至今天才知道是待机CPU超频了，设置408000之后一切正常。建议恢复默认408000。不想超频，请问最大cpu频率设置多少为好呢