stupidloud/nanopi-openwrt #416: Bug: 勾选“禁止解析 IPv6 DNS 记录”会导致无法打开luci管理页面

Sakurakouji-Runa, 2021-03-13:
### 勾选下图中“禁止解析 IPv6 DNS 记录”复选框会导致无法打开luci管理界面，显示为白屏。
[image]

 该bug可以稳定复现，步骤如下：
> 1. 勾选该复选框，保存&应用
> 2. 打开新标签页，访问luci管理界面，显示为白屏
> 3. 取消勾选该复选框，保存&应用
> 4. 刷新在另一标签页的luci管理界面，正常显示

_环境：R2S，2021-03-13版
备注：因该bug导致的无法进入luci管理界面，只在系统启动一分钟左右之后生效，在刚刚启动系统后的约一分钟内可以正常进入luci管理界面。_

klever1988, 2021-03-13:
无法复现，试过主机名和ipv4访问都可以正常打开，也重启过

samanhk2, 2021-03-13:
今天的版本如果是升级过来的会有很多不明bug比如passwall卡死，无速度，IPV6出现故障，但是重新卡刷就好了，我有两个卡，一个卡刷没问题，换另一个升级来的卡问题出现，所以我还是建议今天的版本重新卡刷，比较稳妥。

发自我的iPhone

klever1988, 2021-03-14:
> 今天的版本如果是升级过来的会有很多不明bug比如passwall卡死，无速度，IPV6出现故障，但是重新卡刷就好了，我有两个卡，一个卡刷没问题，换另一个升级来的卡问题出现，所以我还是建议今天的版本重新卡刷，比较稳妥。 发自我的iPhone
> […](#)

昨天比较大的变动就是恢复了upx压缩，还有给autoupdate.sh提供了扩容功能，出问题的哪个你应该是用autoupdate.sh升级的吧？

samanhk2, 2021-03-15:
> 昨天比较大的变动就是恢复了upx压缩，还有给autoupdate.sh提供了扩容功能，出问题的那个你应该是用autoupdate.sh升级的吧？

是的，有用脚本升级

发自我的iPad

Sakurakouji-Runa, 2021-03-18:
> 是的，有用脚本升级 发自我的iPad
> […](#)

感谢回复，更新到最新版本后这个bug就消失了，现在一切正常。问题解决了，我把issue关了。

closed 2021-03-18 by Sakurakouji-Runa