stupidloud/nanopi-openwrt #421: 今天的固件是USB-C的NG R2S起不来三灯全亮 MicroUSB 的OK

bg4drl, 2021-03-15:
今天的固件是USB-C的R2S起不来 MicroUSB OK
同样的R2S 新款USB-C和老款MicroUSB供电的 硬件可能有区别
请大佬针对这点 找出else汇通办法.谢谢

klever1988, 2021-03-15:
无法启动  三灯全亮？ 我不知道usb-c款硬件有多大区别 感觉应该是差不多的
你用同样的卡 同样的电源头试试两台机

bg4drl, 2021-03-15:
是的 用usb转micro的,  
试过了 多启动几次 typeC的起来了
两个R2S都好了. 
卡换了几张同类的32GB的 ,感觉是有点挑卡 
但不是绝对....
R4S的4GB一直没这类事情,
重来没有因卡或者电源匹配问题而起不来. 
今天三个设备两个不同时期的R2S和R4S的报告如上
谢谢大佬来回信

klever1988, 2021-03-15:
对的r2s很挑卡 可能是硬件原因
如果是三等全亮 但是电脑上本地连接又不能识别网络
很可能是启动起来了 但是因为供电不足 固件又超频导致死机的 考虑弄个c口的电源头 弄个好的双头c线供电试试

bg4drl, 2021-03-15:
谢谢大佬回信
给出的见解,
本人是精品主义者 
基本用的都是有印良品 哈哈 
非常感谢 百忙之中的回复.

klever1988, 2021-03-15:
不是酱紫的，现在品牌充电头大多是QC、PD快充协议，但是r2s这种老硬件（包括树莓派）根本不支持这些协议，它只能在5v电压下要求电流。品牌充电头不一定在5v段有很好供电能力，多数是额定1A，电流再大就会出现压降。反而在5V段标称有2/3A供电能力的电源头，都是某些OEM甚至杂牌子的。我的建议就是买这种，而且最好是线和头连在一起的，电阻最小