stupidloud/nanopi-openwrt #423: 在线更新怎么不能用了?

Goodman-vip, 2021-03-16:
终端内在线升级方法：
wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh
(脚本由gary lau提供，非常感谢！)

Goodman-vip, 2021-03-16:
3-13的版本还能行,在线升到3-14的以后就不能在线升了,不知道什么情况

klever1988, 2021-03-17:
整个升级过程复制一下过来看

Goodman-vip, 2021-03-17:
[image]
一直显示这样

Goodman-vip, 2021-03-17:
不知道什么情况,刚卡刷了试就能用,在线更新后再次更新就不行了,就成这个样的了

klever1988, 2021-03-17:
去掉q，这样子执行看看

wget -O- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh

Goodman-vip, 2021-03-17:
好的.谢谢,回头我试试

Goodman-vip, 2021-03-18:
[image]
卡刷了后,能用一次在线升级,然后今天再升3-18版还是不行,这是按你说的去个q执行的

klever1988, 2021-03-18:
raw.githubusercontent.com被解析到你的本地ip127.0.0.1了，

你开了广告插件的劫持https模式？

klever1988, 2021-03-18:
> [image]
> 卡刷了后,能用一次在线升级,然后今天再升3-18版还是不行,这是按你说的去个q执行的

按q的的目的是让他显示出日志看错误出在哪，不是说按q就有效果

Goodman-vip, 2021-03-18:
没开过,我卡刷了3-16,启动后试了下可以在线更新,更新完了重启后,今天更新又不行了,广告插件从来不用的,不知道什么原因,我是R4S

klever1988, 2021-03-18:
[image]

看这个，域名被解析到127.0.0.1，这一般是广告插件的行为，广告插件对付广告就是把域名转向错误的ip，
不是的话那就要找找原因，或者重新刷了

Goodman-vip, 2021-03-18:
好吧，谢谢了

klever1988, 2021-03-18:
你都启用了什么插件？

Goodman-vip, 2021-03-18:
只启用了PASSWALL

Goodman-vip, 2021-03-18:
我试了其它的,不是PASSWALL的问题,我这测试的是每次卡刷后第一次可以使用在线升级,之后就不能用了,里面的设置都是一样的

klever1988, 2021-03-19:
那就是passwall的设置导致的，你可以关闭passwall看是否能解析到正确的ip上

pp875598763, 2021-03-21:
root@ppserver:~# wget -O- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh
--2021-03-21 19:45:50--  https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh
Resolving github.com... 13.229.188.59
Connecting to github.com|13.229.188.59|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/klever1988/nanopi-openwrt/master/scripts/autoupdate.sh [following]
--2021-03-21 19:45:50--  https://raw.githubusercontent.com/klever1988/nanopi-openwrt/master/scripts/autoupdate.sh
Resolving raw.githubusercontent.com... failed: Name does not resolve.
wget: unable to resolve host address 'raw.githubusercontent.com'
root@ppserver:~# wget -O- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh
--2021-03-21 19:46:31--  https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh
Resolving github.com... 13.229.188.59
Connecting to github.com|13.229.188.59|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/klever1988/nanopi-openwrt/master/scripts/autoupdate.sh [following]
--2021-03-21 19:46:32--  https://raw.githubusercontent.com/klever1988/nanopi-openwrt/master/scripts/autoupdate.sh
Resolving raw.githubusercontent.com... failed: Name does not resolve.
wget: unable to resolve host address 'raw.githubusercontent.com'

应该也是失败了

klever1988, 2021-03-23:
wget: unable to resolve host address 'raw.githubusercontent.com'

pp875598763, 2021-03-24:
> wget: unable to resolve host address 'raw.githubusercontent.com'

好像是passwall这个软件不行 换了ssr可以下载了