stupidloud/nanopi-openwrt #428: 重设wan口的mac地址无效

aimuch, 2021-03-18:
如题，同步的源码有问题，DHDAXCW/NanoPi-R4S-2021也有这个问题

klever1988, 2021-03-18:
不太懂，你是说无法修改mac地址？

aimuch, 2021-03-18:
> 不太懂，你是说无法修改mac地址？

对的

klever1988, 2021-03-18:
可能不支持自己设置mac

aimuch, 2021-04-01:
> 可能不支持自己设置mac
我在网页端修改wan口的mac地址其实是生效了，vim /etc/config/network里看wan_eth0_dev没生效，直接删除然后重启就可以了。

onfig interface 'wan'
        option ifname 'eth0'
        option proto 'dhcp'
        option macaddr 'a0:8c:11:de:5b:11'

config device 'wan_eth0_dev'
        option name 'eth0'
        option macaddr 'fe:5b:6e:63:06:78'

config interface 'wan6'
        option ifname 'eth0'
        option proto 'dhcpv6'

config interface 'vpn0'
        option ifname 'tun0'
        option proto 'none'

DHDAXCW, 2021-04-01:
源码问题，如果把这个删了，就可以修改了。编译前执行 如sed -i '/nanopi-r4s/d' target/linux/rockchip/armv8/base-files/etc/board.d/02_network这种，这个我还没尝试，明天试试
[image]

aimuch, 2021-04-01:
> 源码问题，如果把这个删了，就可以修改了。编译前执行 如sed -i '/nanopi-r4s/d' target/linux/rockchip/armv8/base-files/etc/board.d/02_network这种，这个我还没尝试，明天试试
> [image]

帮

klever1988, 2021-04-03:
> 源码问题，如果把这个删了，就可以修改了。编译前执行 如sed -i '/nanopi-r4s/d' target/linux/rockchip/armv8/base-files/etc/board.d/02_network这种，这个我还没尝试，明天试试
> [image]

但这样就失去了读取网卡原生mac地址的特性了