stupidloud/nanopi-openwrt #439: 在线刷新系统报错

ymai6, 2021-03-21:
--2021-03-21 11:04:13--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/r2s-2021-03-21/md5sum.txt
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-03-21 11:04:17 ERROR 404: NOT FOUND.

镜像下单完成以后，下载md5sum.txt报错，没法继续刷新系统

smilence86, 2021-03-21:
我也遇到这问题

Goodman-vip, 2021-03-21:
都一样的

klever1988, 2021-03-21:
修复了已经 再试试

Goodman-vip, 2021-03-21:
2021-03-21 14:57:50 (716 KB/s) - '/tmp/ddnz.1' saved [17936/17936]

--2021-03-21 14:57:50--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-03-21/r4s.img.gz
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-03-21 14:57:51 ERROR 404: NOT FOUND.

--2021-03-21 14:57:51--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-03-21/r4s.img.md5
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-03-21 14:57:52 ERROR 404: NOT FOUND.

root@ImmortalWrt:~# 
试了，
还是这样

Goodman-vip, 2021-03-21:
2021-03-21 15:21:32 (510 KB/s) - '/tmp/ddnz.5' saved [17936/17936]

--2021-03-21 15:21:32--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-03-21/r4s.img.gz
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-03-21 15:21:34 ERROR 404: NOT FOUND.

--2021-03-21 15:21:34--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-03-21/r4s.img.md5
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-03-21 15:21:36 ERROR 404: NOT FOUND.

root@ImmortalWrt:~#

klever1988, 2021-03-21:
> 2021-03-21 15:21:32 (510 KB/s) - '/tmp/ddnz.5' saved [17936/17936]
>
> --2021-03-21 15:21:32-- https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-03-21/r4s.img.gz [...]

r4s在等几分钟

Goodman-vip, 2021-03-21:
可以了,谢谢