stupidloud/nanopi-openwrt #442: 为啥sd卡始终挂载不了？

lhf19900402, 2021-03-21:
[image]
虽然选中了  ，点保存应用刷新后没反应   在上面已挂载的文件系统里没有显示出来

klever1988, 2021-03-21:
系统日志看看有什么输出

lhf19900402, 2021-03-21:
> 系统日志看看有什么输出

Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.116305] workingset: timestamp_bits=46 max_order=18 bucket_order=0
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.127674] squashfs: version 4.0 (2009/01/31) Phillip Lougher
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.128216] jffs2: version 2.2 (NAND) (SUMMARY) (ZLIB) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.156547] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 252)
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.157229] io scheduler mq-deadline registered
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.164698] dma-pl330 ff1f0000.dmac: Loaded driver for PL330 DMAC-241330
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.165331] dma-pl330 ff1f0000.dmac: 	DBUFF-128x8bytes Num_Chans-8 Num_Peri-20 Num_Events-16
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.169079] Serial: 8250/16550 driver, 4 ports, IRQ sharing enabled
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.171675] printk: console [ttyS2] disabled
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.172202] ff130000.serial: ttyS2 at MMIO 0xff130000 (irq = 12, base_baud = 1500000) is a 16550A
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.173085] printk: console [ttyS2] enabled
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.173878] printk: bootconsole [uart8250] disabled
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.184585] loop: module loaded
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.184884] mtip32xx Version 1.3.1
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.187830] libphy: Fixed MDIO Bus: probed
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.189562] rk_gmac-dwmac ff540000.ethernet: IRQ eth_wake_irq not found
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.190236] rk_gmac-dwmac ff540000.ethernet: IRQ eth_lpi not found
Sun Mar 21 16:52:59 2021 kern.warn kernel: [    0.190947] rk_gmac-dwmac ff540000.ethernet: PTP uses main clock
Sun Mar 21 16:52:59 2021 kern.err kernel: [    0.191526] rk_gmac-dwmac ff540000.ethernet: phy regulator is not available yet, deferred probing
Sun Mar 21 16:52:59 2021 kern.err kernel: [    0.193949] dwc3 ff600000.dwc3: Failed to get clk 'ref': -2
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.195011] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.195592] ehci-platform: EHCI generic platform driver
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.198427] ehci-platform ff5c0000.usb: EHCI Host Controller
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.198964] ehci-platform ff5c0000.usb: new USB bus registered, assigned bus number 1
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.199803] ehci-platform ff5c0000.usb: irq 29, io mem 0xff5c0000
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.213877] ehci-platform ff5c0000.usb: USB 2.0 started, EHCI 1.00
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.215158] hub 1-0:1.0: USB hub found
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.215534] hub 1-0:1.0: 1 port detected
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.216463] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.217052] ohci-platform: OHCI generic platform driver
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.217961] ohci-platform ff5d0000.usb: Generic Platform OHCI controller
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.218590] ohci-platform ff5d0000.usb: new USB bus registered, assigned bus number 2
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.219442] ohci-platform ff5d0000.usb: irq 30, io mem 0xff5d0000
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.282604] hub 2-0:1.0: USB hub found
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.282991] hub 2-0:1.0: 1 port detected
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.284500] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.285017] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 3
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.285971] xhci-hcd xhci-hcd.0.auto: hcc params 0x0220fe64 hci version 0x110 quirks 0x0000000002010010
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.286878] xhci-hcd xhci-hcd.0.auto: irq 164, io mem 0xff600000
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.288402] hub 3-0:1.0: USB hub found
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.288778] hub 3-0:1.0: 1 port detected
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.289593] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.290146] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 4
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.290850] xhci-hcd xhci-hcd.0.auto: Host supports USB 3.0 SuperSpeed
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.291517] usb usb4: We don't know the algorithms for LPM for this host, disabling LPM.
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.292842] hub 4-0:1.0: USB hub found
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.293218] hub 4-0:1.0: 1 port detected
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.294266] usbcore: registered new interface driver usb-storage
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.294986] i2c /dev entries driver
Sun Mar 21 16:52:59 2021 kern.info kernel: [    0.298052] rk808 1-0018: chip id: 0x8050
[... cut ...]
Sun Mar 21 16:55:00 2021 cron.err crond[6167]: USER root pid 9657 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun Mar 21 17:00:00 2021 cron.err crond[6167]: USER root pid 10342 cmd /etc/init.d/oled restart >/dev/null 2>&1
Sun Mar 21 17:02:10 2021 daemon.err block: /dev/mmcblk0p1 is already mounted on /mnt/mmcblk0p1
Sun Mar 21 17:02:10 2021 daemon.err block: /dev/mmcblk0p2 is already mounted on /
Sun Mar 21 17:02:10 2021 daemon.err block: No "mount.exfat" utility available
Sun Mar 21 17:02:10 2021 daemon.err block: mounting /dev/sda1 (exfat) as /mnt/sda1 failed (2) - No such file or directory

lhf19900402, 2021-03-21:
> 系统日志看看有什么输出

exfat格式的盘貌似无法识别  我改成fat的能识别了

klever1988, 2021-03-21:
Sun Mar 21 17:02:10 2021 daemon.err block: No "mount.exfat" utility available

看这个提示是没有exfat的用户空间工具支持，没办法，还是用ntfs或者fat吧

sunseago, 2021-03-21:
> Sun Mar 21 17:02:10 2021 daemon.err block: No "mount.exfat" utility available
> 
> 看这个提示是没有exfat的用户空间工具支持，没办法，还是用ntfs或者fat吧

老版本是可以挂载exfat的，新的不行

lhf19900402, 2021-03-21:
> ntfs或者fat

ntfs只有windows支持   fat最大文件限制太小    exfat最合适。。。。

klever1988, 2021-03-21:
> > ntfs或者fat
> 
> ntfs只有windows支持 fat最大文件限制太小 exfat最合适。。。。

ntfs是可以在固件里面挂载的，

fat我试过也不行

lhf19900402, 2021-03-22:
> > > ntfs或者fat
> > 
> > 
> > ntfs只有windows支持 fat最大文件限制太小 exfat最合适。。。。
> 
> ntfs是可以在固件里面挂载的，
> 
> fat我试过也不行

FAT32的内存卡通过读卡器插优盘我这边是可以的     NTFS对用macos的人来说不太友好😔    大神看看你能不能优化一下  支持exfat格式的U盘👍

sunseago, 2021-03-22:
> > > > ntfs或者fat
> > > 
> > > 
> > > ntfs只有windows支持 fat最大文件限制太小 exfat最合适。。。。
> > 
> > 
> > ntfs是可以在固件里面挂载的，
> > fat我试过也不行
> 
> FAT32的内存卡通过读卡器插优盘我这边是可以的 NTFS对用macos的人来说不太友好😔 大神看看你能不能优化一下 支持exfat格式的U盘👍

是的，赞同

klever1988, 2021-03-22:
exfat可以了