stupidloud/nanopi-openwrt #443: SMB用root账户始终登不上

lhf19900402, 2021-03-21:
[image]
[image]

去年以前的固件可以的   友善自己的系统也可以的   现在就死活登录不上

klever1988, 2021-03-21:
luci-app-cifsd依赖ksmbd需要内核模块支持，slim后装的插件可能不行
你试一下luci-app-samba这种

klever1988, 2021-03-21:
试过luci-app-samba4可以

Nonosword, 2021-03-21:
奇怪了，ksmbd版的，怎么配置密码

lhf19900402, 2021-03-21:
> 试过luci-app-samba4可以

👌

Nonosword, 2021-03-21:
> 奇怪了，ksmbd版的，怎么配置密码

root@ImmortalWrt:~# ksmbd.adduser -a root
-bash: ksmbd.adduser: command not found

klever1988, 2021-03-21:
> > 奇怪了，ksmbd版的，怎么配置密码
> 
> root@ImmortalWrt:~# ksmbd.adduser -a root
> -bash: ksmbd.adduser: command not found

[image]

你得装一下这个软件，adduser命令才能用

Nonosword, 2021-03-21:
> > > 奇怪了，ksmbd版的，怎么配置密码
> > 
> > 
> > root@ImmortalWrt:~# ksmbd.adduser -a root
> > -bash: ksmbd.adduser: command not found
> 
> [image]
> 
> 你得装一下这个软件，adduser命令才能用

[image]
我用的20号的R4固件，我看你r4等的源里确实没有ksmbd-tools和ksmbd-utils，你这是哪个源里的呀

Nonosword, 2021-03-21:
> > > > 奇怪了，ksmbd版的，怎么配置密码
> > > 
> > > 
> > > root@ImmortalWrt:~# ksmbd.adduser -a root
> > > -bash: ksmbd.adduser: command not found
> > 
> > 
> > [image]
> > 你得装一下这个软件，adduser命令才能用
> 
> [image]
> 我用的20号的R4固件，我看你r4等的源里确实没有ksmbd-tools和ksmbd-utils，你这是哪个源里的呀

还有个问题，opkg界面的“刷新”按钮总是消失，得重启才能恢复
[image]

klever1988, 2021-03-21:
[image]

改成snapshots应该就有

软件源列表刷新一次就生成了缓存，重启之前这份缓存保持存在，所以luci界面不让你再次刷新，只能进终端执行opkg update了

Nonosword, 2021-03-22:
> [image]
> 
> 改成snapshots应该就有
> 
> 软件源列表刷新一次就生成了缓存，重启之前这份缓存保持存在，所以luci界面不让你再次刷新，只能进终端执行opkg update了
[image]
装了ksmbd-utils后也不是很好用，最后还是用samba4解决了，就是samba4的界面，输入框尺寸没有适配，不过不影响食用。

Neustradamus, 2021-09-17:
Any news?
It is solved?