stupidloud/nanopi-openwrt #446: 最新版（R1S-2021-03-21）广告过滤大师plus(adbyby plus)无法启动

Hank-Zhu, 2021-03-22:
最新版（R1S-2021-03-21）的广告过滤大师plus(adbyby plus)无法启动，进后台手动执行了一下，貌似编译出来的文件有错误，请大神有时间的话看一下。

另外也下了03-19的img也有同样的问题。

klever1988, 2021-03-22:
[image]

可以执行啊

Hank-Zhu, 2021-03-23:
@klever1988  大神你好
页面启动adbyby一直显示未运行
[image]
连上ssh执行adbyby显示： cannot execute binary file: Exec format error
[image]
我用的是R1S-H5，刷完镜像后除了联网未改动任何设置，之前我用的是R1S-Lienol-2020-12-01，adbyby没有问题。是否可以传一份编译好的adbyby给我再试一下？

谢谢

klever1988, 2021-03-26:
这个玩意只有bin提供 http://www.adbyby.com/
不是从源文件编译的 不能运行的话就没办法了