stupidloud/nanopi-openwrt #448: R1S-H3 no wifi

Dagelinckx, 2021-03-22:
I'm new here and I flashed the latest version of the R1S-H3 but under the network tab I'm not able to find any settings for the build-in wifi??

13311031900, 2021-03-22:
我放在R1上用WIFI没问题,固件正常.

Dagelinckx, 2021-03-22:
I have installed the R1S-H3 version but it shows as NanoPi R1 and no wifi settings under the network tab, under interfaces no wifi to be seen as well, any suggestions??

drw2005, 2021-03-23:
在我的NanoPi R1上刷了R1S-H3 slim版本，没有任何问题。

klever1988, 2021-03-24:
> I have installed the R1S-H3 version but it shows as NanoPi R1 and no wifi settings under the network tab, under interfaces no wifi to be seen as well, any suggestions??

is it fixed in the later released firmware? I have add usb-modeswitch to it.

Dagelinckx, 2021-03-25:
I have the friedlyarm R1S-H3 and I installed the latest version of March 21 but no wifi under the network tab, I as well bought the recommended comfast usb and as well nothing under network tab, I installed original version of friendlyarm and wifi is there and when I input the comfast usb it works, sure hardware is no problem but not sure why it not works with this software version??

dannywen2, 2021-04-01:
同样是此问题,r1s h3
希望大佬解决下