stupidloud/nanopi-openwrt #449: 在线升级脚本出错

ymai6, 2021-03-22:
--2021-03-22 20:32:50--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-03-22/r2s.img.gz
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 110788394 (106M) [application/octet-stream]
Saving to: 'r2s.img.gz'

r2s.img.gz                                        100%[=============================================================================================================>] 105.66M   819KB/s    in 30s

2021-03-22 20:33:26 (3.57 MB/s) - 'r2s.img.gz' saved [110788394/110788394]

今天的固件还没更新，尝试下载昨天的固件
--2021-03-22 20:33:26--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-03-21/r2s.img.gz
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 110747468 (106M) [application/octet-stream]
Saving to: 'r2s.img.gz'

r2s.img.gz

ymai6, 2021-03-22:
r2s今天的固件（3.22日）下载完成以后，报：今天的固件还没更新，尝试下载昨天的固件。开始重新下载昨天的固件

klever1988, 2021-03-22:
判断写反了。。。 再试试

Goodman-vip, 2021-03-23:
[image]
还是今天没有的情况下无法自动更新昨天的