stupidloud/nanopi-openwrt #451: 希望能更新一下shadowsocksr plus+，谢谢

fayewonghz, 2021-03-22:
希望能更新一下shadowsocksr plus+，谢谢

klever1988, 2021-03-22:
都是183.9啊 有落后吗？

smilence86, 2021-03-23:
我试过3月16号、20号r2s版本，感觉有问题，卡刷导入备份（来自以前lean版），无法启动ssrp，helloworld也有问题，于是用passwall，发现内存占用太高，目前第3天内存还剩13%：
[image]

klever1988, 2021-03-23:
> 我试过3月16号、20号r2s版本，感觉有问题，卡刷导入备份（来自以前lean版），无法启动ssrp，helloworld也有问题，于是用passwall，发现内存占用太高，目前第3天内存还剩13%：
> [image]

以前lean版的ssrp插件和现在相比变化很大，配置文件的格式也不一样了，跨度太大，直接导出导入肯定不能用。最好是重新配置。
负载数字（正常在零点几）和内存占用都不正常，要进入终端看看什么进程造成的死循环。

smilence86, 2021-03-23:
我没在家，frp只穿透了80端口，自带的TTYD无法用，又不敢再去配一个22端口，怕死机，现在又自动恢复正常了：
[image]

不过passwall内存占用还是比ssrp高，以前使用ssrp内存剩余64%，开了相同的其他插件，所以我还是希望继续用ssrp。

smilence86, 2021-03-23:
刚才在线刷了最新版3月23日r2s，删了所有节点重新订阅，现在ssrp能正常使用，但重启后不能自动代理，需要手动保存一次才行，不知道是不是哪个配置还需要删了。

closed 2021-03-23 by klever1988