stupidloud/nanopi-openwrt #458: opkg 安装kmod-l2tp报错-请问有没有编译好的ipk呀

jason416, 2021-03-24:
用3-23版本，安装l2tp支持时，会有kmod报错，无法安装的提供。
[image]

klever1988, 2021-03-26:
[image]

可以正常安装了

[image]

jason416, 2021-03-27:
> [image]
> 
> 可以正常安装了
> 
> [image]

谢谢大佬恢复，请问我需要如何操作呢？我目前用的是3-23版本。

klever1988, 2021-03-27:
下载新版呀

jason416, 2021-03-27:
> 下载新版呀

我用了3-27号得，发现使用不了了呢。有如下报错：

> Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: start_pppd: I'm running:
Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: "/usr/sbin/pppd"
Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: "plugin"
Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: "pppol2tp.so"
Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: "pppol2tp"
Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: "8"
Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: "passive"
Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: "nodetach"
Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: ":"
Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: "file"
Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: "/tmp/l2tp/options.vpn0"
Sat Mar 27 14:37:05 2021 daemon.err pppd[6352]: Error loading shared library /usr/lib/pppd/2.4.8/pppol2tp.so: No such file or directory
Sat Mar 27 14:37:05 2021 daemon.err pppd[6352]: Couldn't load plugin pppol2tp.so
Sat Mar 27 14:37:05 2021 daemon.debug xl2tpd[1872]: child_handler : pppd exited for call 2730 with code 2
Sat Mar 27 14:37:05 2021 daemon.info xl2tpd[1872]: call_close: Call 46639 to x.x.x.x disconnected

我看了下，确实没有pppol2tp.so文件。
```
root@ImmortalWrt:~# ls /usr/lib/pppd/2.4.8/
rp-pppoe.so
root@ImmortalWrt:~# 
```

klever1988, 2021-03-27:
[image]

我用slim版测试的 你用大杂烩么？

jason416, 2021-03-28:
@klever1988 对，我用的大杂烩版本，就有上面的问题~

jason416, 2021-04-08:
用slim版本，本地仓库直接安装可以解决。

closed 2021-04-08 by jason416