stupidloud/nanopi-openwrt #459: 不能从之前的固件直接终端在线升级到slim版么？

nonozone, 2021-03-24:
求问，命令没反应

klever1988, 2021-03-24:
[image]

这个不是吗

nonozone, 2021-03-24:
> [image]
> 
> 这个不是吗

没反应，事实上两个命令都没反应...

试了下，好像是无法从路由器直接下载这个脚本

[image]

nonozone, 2021-03-25:
关掉所有广告过滤，然后重启之后可以了。

BTW问一句，在新版本搜索下载ssrplus ，为啥不行？

[image]

klever1988, 2021-03-25:
因为上游把一些插件代码从base移到packages，导致编译生成的本地软件源收录不完整，在修复中~~

nonozone, 2021-03-25:
还有个问题，更新过来之后，好像DNS不生效，网络通了，域名都ping不通，lan 都设置了，不知道哪里的问题

nonozone, 2021-03-25:
最终还是直接卡刷解决了...

nonozone, 2021-03-25:
> 因为上游把一些插件代码从base移到packages，导致编译生成的本地软件源收录不完整，在修复中~~

那现在哪个插件可以正常用的？好急- - 哈哈

klever1988, 2021-03-25:
> > 因为上游把一些插件代码从base移到packages，导致编译生成的本地软件源收录不完整，在修复中~~
> 
> 那现在哪个插件可以正常用的？好急- - 哈哈

修复了 试试

nonozone, 2021-03-26:
好像还是不行，重启也不行，需要升级固件么？

nonozone, 2021-03-26:
更新固件后解决，谢谢。

closed 2021-03-26 by nonozone