stupidloud/nanopi-openwrt #462: 【PP】smartdns 安装后没有管理界面？

pp875598763, 2021-03-25:
重启了也没有

klever1988, 2021-03-25:
你装的是luci-app-smartdns吧？ 切换其他主题再切换回来试试

pp875598763, 2021-03-25:
> 你装的是luci-app-smartdns吧？ 切换其他主题再切换回来试试

试过了 还是没有

klever1988, 2021-03-25:
[image]
[image]

没毛病啊

pp875598763, 2021-03-26:
[image]
[image]
[image]
你看

wei727, 2021-04-05:
> [image]
> [image]
> 
> 没毛病啊

我也有同样问题，装上了服务里面不显示。切换主题和重启都没用！
[image]
[image]

wei727, 2021-04-05:
> [image]
> [image]
> [image]
> 你看

https://github.com/pymumu/smartdns/releases
官网去下安装包自己装吧很简单的，我也是在线安装不起自己离线安装重启就有了。不要下错安装包就好！
smartdns.1.2020.09.08-2235.arm-openwrt-all
luci-app-smartdns.1.2020.09.08-2235.all-luci-compat-all
下载这两个安装包重启就有了。
[image]

pp875598763, 2021-04-05:
> > [image]
> > [image]
> > [image]
> > 你看
> 
> https://github.com/pymumu/smartdns/releases
> 官网去下安装包自己装吧很简单的，我也是在线安装不起自己离线安装重启就有了。不要下错安装包就好！
> smartdns.1.2020.09.08-2235.arm-openwrt-all
> luci-app-smartdns.1.2020.09.08-2235.all-luci-compat-all
> 下载这两个安装包重启就有了。
> [image]

感谢 可以了

SeaRat, 2024-07-28:
原来要安装 compat 包啊，可是我的版本是21.02.1 且安装教程里不是说 19.07 之前的版本才需要吗？