stupidloud/nanopi-openwrt #464: 请问nanopi neo3可以用这款固件吗？

thaoeu, 2021-03-25:
和R2s同款芯片的，打算试一下。

lyricusark, 2021-03-25:
同NEO3，刚试了下，原IP打不开网页控制界面。路由器识别了个immortalwrt，物理地址和ip都变了，尝试用新IP也无法访问。

klever1988, 2021-03-25:
> 同NEO3，刚试了下，原IP打不开网页控制界面。路由器识别了个immortalwrt，物理地址和ip都变了，尝试用新IP也无法访问。

什么原IP，还原备份的？还是你用autoupdate刷的

lyricusark, 2021-03-26:
> 
> 
> > 同NEO3，刚试了下，原IP打不开网页控制界面。路由器识别了个immortalwrt，物理地址和ip都变了，尝试用新IP也无法访问。
> 
> 什么原IP，还原备份的？还是你用autoupdate刷的

之前刷的minimal做旁路由用，主路由给neo3绑定了个固定IP。这次直接重新刷slim，原来IP登录不了，主路由上看到neo3的物理地址发生变化，给重新分配了个ip地址，尝试用新IP也无法登录（才看见可以直接 http://immortalwrt/ 登录，我当时没试过），换回原来的系统后一切正常。如果确定neo3可以用slim的话，我下来再鼓捣鼓捣。

msonicc, 2021-04-04:
要USB外接网卡吧，原生的网号是WAN口 USB 网卡是 LAN口？

lyricusark, 2021-04-05:
> 
> 
> 要USB外接网卡吧，原生的网号是WAN口 USB 网卡是 LAN口？

或许是这个原用吧，反正这个版本我的NEO3是连不上了。之前还是minimal的时候，就算刷了R2S的系统，装NEO3后会自动识别只有一个wan口。

thaoeu, 2021-04-05:
> 要USB外接网卡吧，原生的网号是WAN口 USB 网卡是 LAN口？
确实可行，试过别的固件可以，回去试试这款，不知道这款对wifi的支持怎么样，买了接在neo3 2.0口上的wifi拓展。