stupidloud/nanopi-openwrt #475: r2s0328不能在线升级

symantes, 2021-03-27:
r2s0328不能在线升级

Goodman-vip, 2021-03-28:
嗯,我今天也不能在线升级了,一样的提示

klever1988, 2021-03-28:
拍完整，右侧有一段url没看到

symantes, 2021-03-28:
root@ImmortalWrt:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh
Downloading https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
--2021-03-28 13:48:20--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/raw/zstd-bin/truncate
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 403 FORBIDDEN
2021-03-28 13:48:22 ERROR 403: FORBIDDEN.

pprice2018, 2021-03-28:
一样的提示

klever1988, 2021-03-28:
我改了脚本，你再试一下

Goodman-vip, 2021-03-28:
能行了

pprice2018, 2021-03-28:
ok正常了，感谢大佬的更新！！

symantes, 2021-03-28:
r4s在线升级报错：
root@ImmortalWrt:~# wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh
Downloading https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
Installing resize2fs (1.45.6-2) to root...
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base/resize2fs_1.45.6-2_aarch64_generic.ipk
Configuring resize2fs.
--2021-03-28 15:11:29--  https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/zstd-bin/truncate
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
Unable to establish SSL connection.

symantes, 2021-03-28:
r2s升级成功！

symantes, 2021-03-28:
谢谢大佬，r4s升级成功！