stupidloud/nanopi-openwrt #477: r1s h5 无法升级

13311031900, 2021-03-28:
没更新，尝试下载昨天的固件
--2021-03-28 20:02:49--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-03-27/
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-03-28 20:02:53 ERROR 404: NOT FOUND.

--2021-03-28 20:02:53--  http://h5.img.gz/
Resolving h5.img.gz... failed: Name does not resolve.
wget: unable to resolve host address 'h5.img.gz'
没找到最新的固件，脚本退出

klever1988, 2021-03-28:
运行的版本是去年的固件吗?

13311031900, 2021-03-28:
> 运行的版本是去年的固件吗?

用的27号版本

13311031900, 2021-03-28:
> 运行的版本是去年的固件吗?

用的这个命令 wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh

13311031900, 2021-03-29:
r1s-h3也是这个问题，无法在线升级

xuealan, 2021-03-29:
文件名不匹配，R1S使用脚本去网站找的是“h5.img.gz”，网站上放的其实是“r1s.img.gz”，所以找不到

13311031900, 2021-03-29:
> 文件名不匹配，R1S使用脚本去网站找的是“h5.img.gz”，网站上放的其实是“r1s.img.gz”，所以找不到

脚本问题

13311031900, 2021-03-29:
> 运行的版本是去年的固件吗?

试了最新版的固件用ttyd执行命令行依然提示找不到固件，索引的固件名不存在
--2021-03-29 12:19:36--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-03-29/
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-03-29 12:19:38 ERROR 404: NOT FOUND.

--2021-03-29 12:19:38--  http://r1.img.gz/
Resolving r1.img.gz... failed: Name does not resolve.
wget: unable to resolve host address 'r1.img.gz'
今天的固件还没更新，尝试下载昨天的固件
--2021-03-29 12:19:39--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-03-28/
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-03-29 12:19:41 ERROR 404: NOT FOUND.

--2021-03-29 12:19:41--  http://r1.img.gz/
Resolving r1.img.gz... failed: Name does not resolve.
wget: unable to resolve host address 'r1.img.gz'
没找到最新的固件，脚本退出
root@ImmortalWrt:~#

klever1988, 2021-03-29:
我更新了，再试试

klever1988, 2021-03-29:
@13311031900 能用吗

13311031900, 2021-03-29:
> @13311031900 能用吗
能用了 多谢！

xuealan, 2021-03-30:
可以了。赞！ 另外请教一下，不知道怎么样可以查看目前正在运行的固件版本？谢谢！