stupidloud/nanopi-openwrt #479: 网络共享Ksmbd服务器配置问题

nonskim, 2021-03-29:
Ksmbd配置有问题，无法注释invalid users = root ，导致win10无法登录网络共享。
求大佬在配置模板中增加。

klever1988, 2021-03-29:
你可以修改这个文件吧/etc/ksmbd/smb.conf.template

nonskim, 2021-03-29:
[image]
可以修改 但是修改后没用  登录界面一直显示 “拒绝访问”

magihub, 2021-03-30:
我也遇到类似情况：若不开启匿名访问，使用root用户不知如何新建密码，使用路由器的管理密码是错误的。我学习过samba共享的密码创建（sampasswd命令），知道即使是使用root用户名登录，登录samba共享的密码和路由器管理密码是两个单独的。但用此sampasswd命令，并不适用于Ksmbd共享，故求大佬告知Ksmbd如何创建root用户的密码？若新建如username的新用户，又如何启用访问？谢谢。

nonskim, 2021-03-31:
> > 续：删除map to guest = Bad User 就可以登录了，但是账户密码不对
> > 使用命令ksmbd.adduser -a root，新建不了用户密码，无效。。。。
> > 实在没办法了，求大佬把ksmbd换成samba吧，这配置太难了
> 
> ksmbd-utils加入了，晚点再试试

命令行有了，但是root用户无法添加
root@ImmortalWrt:~# ksmbd.adduser -a root
[smbuseradd/14640]: ERROR: User name sanity check failure
添加了一个user用户，但是登录还是那个问题，显示“拒绝访问”，删除map to guest = Bad User 也没用。
root@ImmortalWrt:~# ksmbd.adduser -a user
New password:
Retype new password:
[smbuseradd/27362]: INFO: User 'user' added

nonskim, 2021-03-31:
大佬，命令行已经有了，但是root用户添加不了，显示[smbuseradd/14640]: ERROR: User name sanity check failure
其他用户可以添加，但是依然显示“拒绝访问”。map to guest = Bad User已经删除了

klever1988, 2021-03-31:
添加完用户重启试试，可能因为是内核型的软件

nonskim, 2021-03-31:
> 添加完用户重启试试，可能因为是内核型的软件

不行，还是“拒绝访问”。

klever1988, 2021-03-31:
我是一直可以匿名访问，相反的，取消了匿名，创建了账户，仍然可以匿名读写访问。。。:joy:

nonskim, 2021-03-31:
> 我是一直可以匿名访问，相反的，取消了匿名，创建了账户，仍然可以匿名读写访问。。。😂
大佬怎么配置的  具体怎么操作

magihub, 2021-03-31:
> 你用mnt路径试试看

大佬，你还是勾选了允许匿名访问，所以有没有密码都没关系啊，nonskim是想以root用户（或者其它自创用户）加密码的方式进行访问呢。

magihub, 2021-03-31:
另外，我很奇怪的是，为啥我更新了最新的固件，有些设置页面咋都成了英文的呢？明明系统语言选择的是中文啊！
[image]

magihub, 2021-03-31:
还有就是，使用系统-备份时，无法配备AdGurad Home设置页面里的添加的过滤规则、DNS设置等等。

nonskim, 2021-03-31:
@magihub讨不讨厌，乱入蹭热度是怎么回事，有本事解决问题

发自我的iPhone

magihub, 2021-03-31:
我也关心Ksmbd共享的问题啊！因为大佬在此出没，我很想知道结果。加上我遇到的问题，也希望大佬能够注意到。

amaztony, 2021-03-31:
我倒是比较烦恼为什么ksmbd用端口转发不行，转到445端口也不行

amaztony, 2021-03-31:
还想装gowebdav 可以源里没有

nonskim, 2021-04-01:
> [image]
> [image]
> 
> 你用mnt路径试试看

不是 我想看看/etc/ksmbd/smb.conf 以及ksmbd.conf里面的参数

klever1988, 2021-04-01:
[global]
        netbios name = ImmortalWrt
        server string = Ksmbd on OpenWrt
        workgroup = WORKGROUP
        interfaces = br-lan
        bind interfaces only = yes
        ipc timeout = 20
        deadtime = 15
        map to guest = Bad User
        smb2 max read = 64K
        smb2 max write = 64K
        smb2 max trans = 64K
        cache read buffers = no
        cache trans buffers = no

######### Dynamic written config options #########

[mnt]
        path = /mnt
        force user = root
        force group = root
        create mask = 0666
        directory mask = 0777
        browseable = yes
        read only = no
        guest ok = no

我晚上又试了一下，如果一开始就不勾选允许匿名，登录时确实是需要输入账户的，在终端adduser之后即可登录了
ksmbd.adduser -a user -p pass

adduser的时候不要使用系统账户的名字，比如root admin什么的

klever1988, 2021-04-01:
> 另外，我很奇怪的是，为啥我更新了最新的固件，有些设置页面咋都成了英文的呢？明明系统语言选择的是中文啊！
> [image]

这个问题今天的固件修复了， 备份的话是插件自己指定需要备份的内容，和固件没关系啦

magihub, 2021-04-01:
> > 另外，我很奇怪的是，为啥我更新了最新的固件，有些设置页面咋都成了英文的呢？明明系统语言选择的是中文啊！
> 
> 这个问题今天的固件修复了， 备份的话是插件自己指定需要备份的内容，和固件没关系啦

@klever1988 我今天打开昨日的备份文件压缩包，发现果然在etc目录下并不包含agh的相关设置（/etc/AdGuardHome.yaml） 我吃了两次亏 今天后来把 AdGuardHome.yaml 文件也复制到本地了。我不甚了解如何选择设定（用户自定义或者系统预设）备份哪些文件，不过这个yaml文件应该是被忽略掉了，所以才会在恢复后丢失原来已改过的agh设置。包括上面提到的默认的DNS第一个是192.168.1.1，建议直接删掉修改为114.114.114.114和8.8.8.8，我是实测后的体会。
[image]

nonskim, 2021-04-01:
> [global]
> netbios name = ImmortalWrt
> server string = Ksmbd on OpenWrt
> workgroup = WORKGROUP
> interfaces = br-lan
> bind interfaces only = yes
> ipc timeout = 20
> deadtime = 15
> map to guest = Bad User
> smb2 [...]

[image]
[image]
[image]
即使使用一模一样的配置，我也无法登录，我奔溃了。。。

klever1988, 2021-04-01:
> 即使使用一模一样的配置，我也无法登录，我奔溃了。。。

可能是你要共享的目录权限问题，考虑chmod -R 777一下

bingkina, 2021-08-19:
所以这个问题最后算解决了吗？我也遇到这个问题了

edwardfhsiao, 2021-09-13:
有两个问题想问一下:

1, 图一别人有“允许用户”,为什么ksmbd却没有?我现在只能手动改conf文件
[image]
[image]

2, 跟 @amaztony 一样,设置了外网端口转发到445,也不能访问. @amaztony 你最后解决了吗?

edwardfhsiao, 2021-09-13:
贴一下我的配置吧,我有```parant```和```child```两个用户,每次重启都要手动添加加粗部分
<pre>
[global]
	netbios name = ImmortalWrt
	server string = Ksmbd on OpenWrt
	workgroup = WORKGROUP
	interfaces = br-lan
	bind interfaces only = yes
	ipc timeout = 20
	deadtime = 15
	map to guest = Bad User
	smb2 max read = 64K
	smb2 max write = 64K
	smb2 max trans = 64K
	cache read buffers = no
	cache trans buffers = no

######### Dynamic written config options #########

[4T_share]
	path = /mnt/sda2/Share
	create mask = 0666
	directory mask = 0777
	browseable = yes
	read only = yes
	guest ok = no
	<b>read list = parent,child</b>

[4T_root]
	path = /mnt/sda2
	create mask = 0666
	directory mask = 0777
	browseable = yes
	read only = no
	guest ok = no
  	<b>valid users = parent
  	invalid users = child
  	write list = parent
  	writeable = yes</b>
</pre>

Neustradamus, 2021-09-16:
To follow this bug.

edwardfhsiao, 2021-09-17:
Finally, I use OpenWRT over immortalwrt, use samba instead of ksmbd. 
ksmbd seems not able to listen from wan.

Neustradamus, 2021-09-17:
@edwardfhsiao: Can you report your problem here:
- https://github.com/cifsd-team/ksmbd/issues

Note: ksmbd will be integrated Linux Kernel 5.15 (branch currently in test).

edwardfhsiao, 2021-09-17:
> @edwardfhsiao: Can you report your problem here:
> 
> * https://github.com/cifsd-team/ksmbd/issues
> 
> Note: ksmbd will be integrated Linux Kernel 5.15 (branch currently in test).

Sorry, I don't feel good opening another issue since I don't have strong evidence about it.

TMXKWWZ, 2023-12-28:
问题太多了，我这个更尴尬
root@ImmortalWrt:~# ksmbd.adduser -a www -p password
-bash: ksmbd.adduser: command not found

TMXKWWZ, 2023-12-28:
实测，用手机可以正常访问，但是用windowd11是不行的，应该是windows不兼容导致的

2398tr, 2024-06-17:
大佬，ksmbd: fail to load dos attribute. 这个错误怎么解决呢