stupidloud/nanopi-openwrt #480: 无线网卡驱动是不是没有？

siyang814, 2021-03-29:
更新到这个固件无线网卡不能用了， 是否可以添加驱动呢？

klever1988, 2021-03-29:
请尽可能详细描述，设备 固件版本 日期、网卡型号

siyang814, 2021-03-29:
固件是最新的28号的固件， 网卡不知道是什么型号的
[image]

siyang814, 2021-03-29:
是我的网卡有问题吗？ 很久之前刷的一个不知道是什么版本的了， 是可以用的

siyang814, 2021-03-29:
或者支持什么网卡有说明吗？ 我可以再买一个， 如果这个不支持的话

klever1988, 2021-03-29:
你把网卡插入r2s, 然后终端打入dmesg命令  截取最后20行左右的代码过来看看

siyang814, 2021-03-29:
这个是设备的芯片吗  r8152

[    6.946225] usbcore: registered new interface driver r8152
[    6.953785] usbcore: registered new interface driver ums-alauda
[    6.955539] usbcore: registered new interface driver ums-cypress
[    6.957417] usbcore: registered new interface driver ums-datafab
[    6.959061] usbcore: registered new interface driver ums-freecom
[    6.960795] usbcore: registered new interface driver ums-isd200
[    6.962425] usbcore: registered new interface driver ums-jumpshot
[    6.964395] usbcore: registered new interface driver ums-karma
[    6.966768] usbcore: registered new interface driver ums-sddr09
[    6.968510] usbcore: registered new interface driver ums-sddr55
[    6.970242] usbcore: registered new interface driver ums-usbat
[    6.971981] usbcore: registered new interface driver usblp
[    7.023483] xt_time: kernel timezone is -0000
[    7.065964] PPP generic driver version 2.4.2
[    7.067613] PPP MPPE Compression module registered
[    7.070091] NET: Registered protocol family 24
[    7.107020] usb 4-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[    7.143778] r8152 4-1:1.0 (unnamed net_device) (uninitialized): Invalid ether addr 00:00:00:00:00:00
[    7.144616] r8152 4-1:1.0 (unnamed net_device) (uninitialized): Random ether addr a6:c0:21:98:a3:5b
[    7.178857] r8152 4-1:1.0 eth1: v1.10.11
[    7.282103] RTW: module init start
[    7.282425] RTW: rtl8821cu v5.4.1_28754.20180921_COEX20180712-3232
[    7.282965] RTW: build time: Mar 27 2021 03:22:43
[    7.283378] RTW: rtl8821cu BT-Coex version = COEX20180712-3232
[    7.284095] usbcore: registered new interface driver rtl8821cu
[    7.284610] RTW: module init ret=0
[    7.289522] l2tp_ppp: PPPoL2TP kernel driver, V2.0
[    7.293327] kmodloader: done loading kernel modules from /etc/modules.d/*
[    9.786992] xt_FULLCONENAT: RFC3489 Full Cone NAT module
[    9.786992] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
[   11.442995] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[   11.448007] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[   11.448714] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[   11.449568] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[   11.492110] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   11.494124] r8152 4-1:1.0 eth1: carrier on
[   16.574374] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   16.575234] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   36.114927] ksmbd: kill command received
[ 5125.921988] hrtimer: interrupt took 12834 ns

siyang814, 2021-03-29:
列表里面没有无线的选项
[image]

klever1988, 2021-03-29:
r8152是有线网卡

你dmesg没有显示出usb设备插入的迹象啊
正常应该是这样的
[image]

不要带着无线网卡启动，启动完成之后再插入，然后打印dmesg

siyang814, 2021-03-29:
[ 7275.700393] usb 1-1: new high-speed USB device number 2 using ehci-platform
[ 7275.857891] usb-storage 1-1:1.0: USB Mass Storage device detected
[ 7275.859181] scsi host0: usb-storage 1-1:1.0
[ 7276.875516] scsi 0:0:0:0: Direct-Access     Realtek  Driver Storage   1.00 PQ: 0 ANSI: 0 CCS
[ 7276.877741] sd 0:0:0:0: [sda] 16000 512-byte logical blocks: (8.19 MB/7.81 MiB)
[ 7276.881258] sd 0:0:0:0: [sda] Write Protect is off
[ 7276.881689] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[ 7276.884625] sd 0:0:0:0: [sda] No Caching mode page found
[ 7276.885102] sd 0:0:0:0: [sda] Assuming drive cache: write through
[ 7276.915472]  sda:
[ 7276.923266] sd 0:0:0:0: [sda] Attached SCSI removable disk
[ 7276.982845] FAT-fs (sda): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[ 7892.860786] sh (28165): drop_caches: 3
[ 9673.560389] usb 1-1: USB disconnect, device number 2
[ 9801.125310] usb 1-1: new high-speed USB device number 3 using ehci-platform
[ 9801.282581] usb-storage 1-1:1.0: USB Mass Storage device detected
[ 9801.283943] scsi host0: usb-storage 1-1:1.0
[ 9802.316622] scsi 0:0:0:0: Direct-Access     Realtek  Driver Storage   1.00 PQ: 0 ANSI: 0 CCS
[ 9802.318701] sd 0:0:0:0: [sda] 16000 512-byte logical blocks: (8.19 MB/7.81 MiB)
[ 9802.322220] sd 0:0:0:0: [sda] Write Protect is off
[ 9802.322651] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[ 9802.325558] sd 0:0:0:0: [sda] No Caching mode page found
[ 9802.326033] sd 0:0:0:0: [sda] Assuming drive cache: write through
[ 9802.356970]  sda:
[ 9802.364322] sd 0:0:0:0: [sda] Attached SCSI removable disk
[ 9802.422918] FAT-fs (sda): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

siyang814, 2021-03-29:
usb 1-1: USB disconnect, device number 2
[  250.187269] usb 1-1: new high-speed USB device number 3 using ehci-platform
[  250.344756] usb-storage 1-1:1.0: USB Mass Storage device detected
[  250.345995] scsi host0: usb-storage 1-1:1.0
[  251.354517] scsi 0:0:0:0: Direct-Access     Realtek  Driver Storage   1.00 PQ: 0 ANSI: 0 CCS
[  251.356877] sd 0:0:0:0: [sda] 16000 512-byte logical blocks: (8.19 MB/7.81 MiB)
[  251.360360] sd 0:0:0:0: [sda] Write Protect is off
[  251.360788] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[  251.363633] sd 0:0:0:0: [sda] No Caching mode page found
[  251.364108] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  251.394754]  sda:
[  251.402504] sd 0:0:0:0: [sda] Attached SCSI removable disk
[  251.462238] FAT-fs (sda): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

siyang814, 2021-03-29:
好像是把我的网卡识别城了存储设备了？

klever1988, 2021-03-29:
opkg update
opkg install usb-modeswitch

再重新插入试一下

siyang814, 2021-03-29:
感谢大神， 现在可以了， 我的网卡开始工作了， 多谢

MrDecadence, 2021-10-01:
请问2021-9-18版本适合用哪款USB无线网卡？我有看到别人推荐雷凌RT5572芯片的USB无线网卡可以免驱动。