stupidloud/nanopi-openwrt #484: frpc 配置后，访问其他设备正常，但是无法远程访问路由器管理页面

LarryLi-cn, 2021-03-30:
frpc 配置后，tcp协议下，SSH远程可连，远程访问其他设备正常，
LAN口下的另外两个次级路由都可以访问了，
但是无法远程访问主路由R2S的路由器管理页

klever1988, 2021-03-30:
路由器的ssh能连，但是管理页不能，那应该是管理页依赖的uhttpd有什么安全设置阻挡了

LarryLi-cn, 2021-03-30:
我在uhttpd管理页面没找到看起来管用的开关，求指导

klever1988, 2021-03-30:
访问的时候浏览器显示什么提示

LarryLi-cn, 2021-03-30:
This site can’t be reached
The webpage might be temporarily down or it may have moved permanently to a new web address.
ERR_UNSAFE_PORT

LarryLi-cn, 2021-03-30:
frps管理界面看到对应的端口是online的

klever1988, 2021-03-31:
[image]
会不会是这个

LarryLi-cn, 2021-03-31:
我回去试试   印象中可能试过  多谢大佬~

LarryLi-cn, 2021-04-01:
> [image]
> 会不会是这个

我昨晚试了一下还是不管用~  而且我安装了L大的版本也是同样的问题。
另外我看网上有配置这个：option rfc1918_filter '0'
我试了也没用~
好无奈。。。。。。
其他的都能访问只有这路由器管理页面无法访问

klever1988, 2021-04-01:
改用其他办法“回家”咯

LarryLi-cn, 2021-04-02:
> 改用其他办法“回家”咯
唉~  大佬们没遇到过同样的问题吗？

LarryLi-cn, 2021-04-02:
> 改用其他办法“回家”咯
感谢大神的多回复，我终于找到问题了：我设定了一个chrome认为不安全的端口~ 所以一直打不开~  我也换过端口，居然也是一个在chrome 黑名单里的~  真是中大奖了~  
另外附上chrome黑名单端口列表（https://blog.csdn.net/qq_33709508/article/details/108826606），希望大家避免踩坑~
再次感谢大佬的多次回复~

closed 2021-04-02 by LarryLi-cn

klever1988, 2021-04-02:
我才知道chrome还有这么妖的设定。。

麻烦解决 大吉大利