stupidloud/nanopi-openwrt #485: 无线网卡驱动问题

cherish0, 2021-03-30:
使用的网卡为RTL8812BU，设备R4S

友善的官方固件可以驱动

dmesg打印[ 3385.155535] usb 7-1: new high-speed USB device number 2 using xhci-hcd

klever1988, 2021-03-30:
这个驱动没有预置 我添加一下再试试

klever1988, 2021-03-30:
https://github.com/klever1988/nanopi-openwrt/commit/5a9b3a4533fcbafa150492fb9977f9d7d8818123#diff-21bf1f1fda5b924c8b626cdb273e8a11d0506c165b242c4321a42f9b2b8d4e68

已经加入了，可以测试了啊

cherish0, 2021-03-31:
> [5a9b3a4#diff-21bf1f1fda5b924c8b626cdb273e8a11d0506c165b242c4321a42f9b2b8d4e68](https://github.com/klever1988/nanopi-openwrt/commit/5a9b3a4533fcbafa150492fb9977f9d7d8818123#diff-21bf1f1fda5b924c8b626cdb273e8a11d0506c165b242c4321a42f9b2b8d4e68)
> 
> 已经加入了，可以测试了啊

今天（2021-3-31）发布的固件好像没有r4s

cherish0, 2021-03-31:
已测试，无法使用

klever1988, 2021-03-31:
dmesg还是输出一样的信息吗

cherish0, 2021-04-01:
> dmesg还是输出一样的信息吗

```
[102852.098395] xhci-hcd xhci-hcd.0.auto: WARN Cannot submit Set TR Deq Ptr
[102852.098986] xhci-hcd xhci-hcd.0.auto: A Set TR Deq Ptr command is pending.
[102852.120896] RTW: rtw_ndev_uninit(wlan0) if1
[102852.221563] RTW: rtw_dev_unload: driver not in IPS
[102853.616908] usb 7-1: new high-speed USB device number 2 using xhci-hcd
[102853.769188] RTW: [HALMAC]11692M
[102853.769188] HALMAC_MAJOR_VER = 1
[102853.769188] HALMAC_PROTOTYPE_VER = 5
[102853.769188] HALMAC_MINOR_VER = 14
[102853.769188] HALMAC_PATCH_VER = 15
[102853.771669] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[102853.926790] RTW: HW EFUSE
[102853.927049] RTW: 0x000: 29 81 00 3C  09 00 A1 00  B6 04 64 10  00 00 A3 00
[102853.927697] RTW: 0x010: 2F 2F 2F 2F  2F 2F 2A 2A  2A 2A 2A 13  00 00 FF FF
[102853.928344] RTW: 0x020: FF FF 22 22  22 22 22 21  21 20 21 22  23 23 23 23
[102853.929027] RTW: 0x030: 12 00 FF FF  00 FF E0 00  FF FF 34 34  34 35 35 35
[102853.929677] RTW: 0x040: 2F 2F 2F 30  30 13 00 00  FF FF FF FF  25 25 25 25
[102853.930324] RTW: 0x050: 25 25 25 25  25 25 25 25  25 25 0E 00  FF FF 00 FF
[102853.930971] RTW: 0x060: E0 00 FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.931616] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.932262] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.932932] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.933580] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.934227] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  7F 20 1C 00  FF FF FF FF
[102853.934874] RTW: 0x0C0: FF 09 00 11  00 00 00 00  00 FF 03 FF  FF FF FF FF
[102853.935520] RTW: 0x0D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.936167] RTW: 0x0E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.936814] RTW: 0x0F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.937485] RTW: 0x100: DA 0B 12 B8  C1 67 02 E8  4E 06 85 B1  29 09 03 52
[102853.938132] RTW: 0x110: 65 61 6C 74  65 6B 1F 03  55 53 42 33  2E 30 20 38
[102853.938779] RTW: 0x120: 30 32 2E 31  31 61 63 20  31 32 30 30  4D 20 41 64
[102853.939426] RTW: 0x130: 61 70 74 65  72 08 03 31  32 33 34 35  36 FF FF FF
[102853.940071] RTW: 0x140: 33 0A 1F 01  00 00 21 0F  FF FF FF FF  FF FF FF FF
[102853.940718] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.941388] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.942037] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.942684] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.943331] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.943976] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.944623] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.945291] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.945939] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.946586] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.947233] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.947879] RTW: 0x200: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.948526] RTW: 0x210: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.949197] RTW: 0x220: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.949845] RTW: 0x230: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.950492] RTW: 0x240: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.951139] RTW: 0x250: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.951785] RTW: 0x260: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.952432] RTW: 0x270: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.953168] RTW: 0x280: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.953894] RTW: 0x290: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.954536] RTW: 0x2A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.955178] RTW: 0x2B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.955819] RTW: 0x2C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.956461] RTW: 0x2D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.957129] RTW: 0x2E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.957772] RTW: 0x2F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[102853.958449] RTW: hal_com_config_channel_plan chplan:0x7F
[102853.958931] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[102853.965349] RTW: [HALMAC][ALWAYS]shall R reg twice!!
[102854.009158] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[102854.009742] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[102854.010302] RTW: ERROR [HALMAC][ERR]Dump efuse in suspend
[102854.011496] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[102854.013879] RTW: rtw_ndev_init(wlan0) if1 mac_addr=e8:4e:06:85:b1:29
[102854.015521] RTW: rtw_ndev_init(wlan1) if2 mac_addr=ea:4e:06:85:b1:29

```

网络设置里能够显示无线的选项，但是无论怎么配置，都显示”无线未开启或未关联“，无法启用ap模式。

klever1988, 2021-04-01:
这是未知也是无解的问题，在r1s/r2s上也是如此
你可以使用这个脚本设置wifi
https://github.com/klever1988/nanopi-openwrt/blob/master/usb-wifi.sh