stupidloud/nanopi-openwrt #486: 有个问题，关于DNS，发现家里的设备，每次打开某些网站/app（不固定）的时候，总是有时候有些图有些页面就突然好久刷不出来

nonozone, 2021-03-30:
有个问题，关于DNS，发现家里的设备，每次打开某些网站/app（不固定）的时候，总是有时候有些图有些页面就突然好久刷不出来，要过十来秒才能出来。

是不是哪里设置不对，不要DNS重定向，或者不要开Dnsmasq ？

Jecvay, 2021-03-30:
尝试下禁用 chrome 的 quic 功能

nonozone, 2021-03-30:
> 尝试下禁用 chrome 的 quic 功能

其实电脑上还不明显，主要是手机上比较明显，并且不是一部手机的问题。

nonozone, 2021-03-30:
测试了下，有可能跟ssr+的设置有关系，把ssr+里面的DNS解析方式改成使用本机端口*****的DNS服务好像好一点，再观察观察吧...

nonozone, 2021-03-30:
> 测试了下，有可能跟ssr+的设置有关系，把ssr+里面的DNS解析方式改成使用本机端口*****的DNS服务好像好一点，再观察观察吧...

好像不是这个原因...

Chocolateyk, 2021-03-30:
我有类似的问题。我这边的问题是一段时间没接入的设备（我这边都是无线接入）重新接入后打开网页（无论墙内外）会非常慢，最差情况是打不开，最好情况是图片加载很慢，过会儿就会自己慢慢恢复。只开了留学，改了lan口ip，其他几乎没动啥设置，换到别的固件就没问题。

klever1988, 2021-03-30:
turboacc的dns加速打开就可以大幅度降低延迟了

nonozone, 2021-03-31:
> turboacc的dns加速打开就可以大幅度降低延迟了

这个已经有了。

dalidaji, 2021-03-31:
尝试关闭下去广告插件，尤其是过滤DNS的去广告插件。

nonozone, 2021-03-31:
> 尝试关闭下去广告插件，尤其是过滤DNS的去广告插件。

暂时路由器没有启用广告过滤。

klever1988, 2021-03-31:
是不是上游dns选择不佳， 换一批试试看

closed 2021-04-01 by nonozone