stupidloud/nanopi-openwrt #490: R1P是什么版本?

13311031900, 2021-03-30:
R1P是什么版本?

klever1988, 2021-03-30:
香橙派 r1 plus

buos1984, 2021-04-03:
硬件配置和R2S一模一样，固件可以通刷，而且目前没有被炒作出天价，推荐香橙派 r1 plus

duaneya, 2021-04-09:
> 硬件配置和 R2S 一模一样，固件可以通刷，而且目前没有被炒作出天价，推荐香橙派 r1 plus
我买了，唉，只有官方提供的固件可以开机，别的都不行

buos1984, 2021-04-09:
> > 硬件配置和 R2S 一模一样，固件可以通刷，而且目前没有被炒作出天价，推荐香橙派 r1 plus
> > 我买了，唉，只有官方提供的固件可以开机，别的都不行

1988大佬已经适配了r1p，直接刷就是。

duaneya, 2021-04-09:
[klever1988/nanopi-openwrt] R1P是什么版本? (#490)

"target": "https://github.com/klever1988/nanopi-openwrt/issues/490#issuecomment-816598364",
"url": "https://github.com/klever1988/nanopi-openwrt/issues/490#issuecomment-816598364",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]