stupidloud/nanopi-openwrt #491: 3.31 r2s镜像在线升级报错

ymai6, 2021-03-31:
如下：

The partition table has been altered.
Syncing disks.
解压已完成，准备编辑镜像文件，写入备份信息
sort: standard output: No space left on device
sh: write error: No space left on device
sh: write error: No space left on device
Saving config files...
Failed to create the configuration backup.
root@ImmortalWrt:~#

hwdef, 2021-03-31:
遇到同样错误，r2s slim 版本

[image]

klever1988, 2021-03-31:
更新了，加了一个10秒的延时，应该可以了

hwdef, 2021-03-31:
[image]

还是会报 Failed to create the configuration backup.

klever1988, 2021-04-01:
奇怪。。 重启一下再试试

nonozone, 2021-04-01:
> 奇怪。。 重启一下再试试

一样的问题，重启多次都还存在。

hwdef, 2021-04-06:
4月5日的slim依旧有同样的问题。

hwdef, 2021-04-06:
[image]

klever1988, 2021-04-06:
复制一下完整的过程输出吧

hwdef, 2021-04-07:
使用4.7版本的升级成功了，但是ssrp丢失了。

hwdef, 2021-04-07:
[image]

重装后报如下错误，应该是与之前的配置文件冲突了。