stupidloud/nanopi-openwrt #496: 求助。。R2S终端内用在线升级脚本没反应

maiapig, 2021-03-31:
R2S终端内用在线升级脚本没反应，固件版本是ImmortalWrt 18.06-SNAPSHOT r9281+14-671443066 / LuCI openwrt-18.06 branch (git-21.060.52352-74ea004)，求大佬解答，万分感谢
[image]

dalidaji, 2021-04-01:
wget -qO- https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/master/scripts/autoupdate.sh | sh

maiapig, 2021-04-02:
> wget -qO- https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/master/scripts/autoupdate.sh | sh

。。确实可以了 看来是被墙了是吧 有什么办法可以让这个脚本走ssrplus+的代理吗

klever1988, 2021-04-02:
github.io
github.com
githubapp.com
githubassets.com
githubusercontent.com
添加到“强制”

maiapig, 2021-04-04:
> github.io
> github.com
> githubapp.com
> githubassets.com
> githubusercontent.com
> 添加到“强制”

啊这。。现在ssrp是英文版的了，“强制”是哪个选项啊，谢谢大佬
[image]

TomFoxLee, 2021-04-05:
> 
> 
> > github.io
> > github.com
> > githubapp.com
> > githubassets.com
> > githubusercontent.com
> > 添加到“强制”
> 
> 啊这。。现在ssrp是英文版的了，“强制”是哪个选项啊，谢谢大佬

應該是 bypass domain list 那一塊。

另外你可以試試 "opkg update luci-i18n-ssr-plus-zh-cn" 看能不能把中文介面裝回來。。。

closed 2021-04-17 by maiapig