stupidloud/nanopi-openwrt #506: 请问带佬怎么自己扩容分区啊？

baby195lxl, 2021-04-02:
全插件的很多不太需要，安装silm版想自己装插件但安装几个之后就没有多少容量了，查了资料也没有找到怎么处理
[image]

klever1988, 2021-04-02:
卡拔下来用winddows+读卡器 
软件可以选择diskgenius

baby195lxl, 2021-04-02:
好的，谢谢

baby195lxl, 2021-04-03:
> 卡拔下来用winddows+读卡器
> 软件可以选择diskgenius
带佬，用diskgenius说"block bitmap中有标记为已使用的未用簇",用chkdsk报错"文件系统的类型是RAW，CHKDSK无法供RAW驱动器使用」"，这个和内存卡有关系吗

closed 2021-04-03 by baby195lxl

caseysui, 2021-04-04:
> block bitmap中有标记为已使用的未用簇

你怎么解决的，我也遇到了这个问题。。。

baby195lxl, 2021-04-04:
虚拟机安装Linux使用gparted

> > block bitmap中有标记为已使用的未用簇
> 
> 你怎么解决的，我也遇到了这个问题。。。离线扩容提示 block bitmap中有标记为已使用的未用簇

klever1988, 2021-04-08:
> 虚拟机安装Linux使用gparted
> 
> > > block bitmap中有标记为已使用的未用簇
> > 
> > 
> > 你怎么解决的，我也遇到了这个问题。。。离线扩容提示 block bitmap中有标记为已使用的未用簇

其实用读卡器插到运行中openwrt也可以用cfdisk和resize2fs扩容，原先不知道每个用户linux基础怎么样，我都不知道怎么回答，只能从简

baby195lxl, 2021-04-08:
> > 虚拟机安装Linux使用gparted
> > > > block bitmap中有标记为已使用的未用簇
> > > 
> > > 
> > > 你怎么解决的，我也遇到了这个问题。。。离线扩容提示 block bitmap中有标记为已使用的未用簇
> 
> 其实用读卡器插到运行中openwrt也可以用cfdisk和resize2fs扩容，原先不知道每个用户linux基础怎么样，我都不知道怎么回答，只能从简

谢谢带佬了