stupidloud/nanopi-openwrt #508: 【PP】今天自动更新r2s后不能上网了 lan口被禁用的样子

pp875598763, 2021-04-03:
分配不到ip lan口不闪
重新刷机上周的版本后 恢复正常

pp875598763, 2021-04-03:
root@ppserver:~# wget -O- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh
--2021-04-03 16:48:59--  https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh
Resolving github.com... 13.229.188.59
Connecting to github.com|13.229.188.59|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/klever1988/nanopi-openwrt/master/scripts/autoupdate.sh [following]
--2021-04-03 16:49:00--  https://raw.githubusercontent.com/klever1988/nanopi-openwrt/master/scripts/autoupdate.sh
Resolving raw.githubusercontent.com... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3329 (3.3K) [text/plain]
Saving to: 'STDOUT'

-                                               100%[======================================================================================================>]   3.25K  --.-KB/s    in 0s      

2021-04-03 16:49:01 (7.58 MB/s) - written to stdout [3329/3329]

Downloading https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_core
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_base
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_luci
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_routing
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_telephony
--2021-04-03 16:49:06--  https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/zstd-bin/truncate
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 45177 (44K) [application/octet-stream]
Saving to: '/tmp/truncate'

truncate                                        100%[======================================================================================================>]  44.12K  --.-KB/s    in 0.06s   

2021-04-03 16:49:07 (752 KB/s) - '/tmp/truncate' saved [45177/45177]

--2021-04-03 16:49:07--  https://ghproxy.com/https://raw.githubusercontent.com/klever1988/nanopi-openwrt/zstd-bin/ddnz
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 17936 (18K) [application/octet-stream]
Saving to: '/tmp/ddnz'

ddnz                                            100%[======================================================================================================>]  17.52K  --.-KB/s    in 0.008s  

2021-04-03 16:49:09 (2.12 MB/s) - '/tmp/ddnz' saved [17936/17936]

--2021-04-03 16:49:09--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-04-03/r2s.img.gz
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-04-03 16:49:10 ERROR 404: NOT FOUND.

今天的固件还没更新，尝试下载昨天的固件
--2021-04-03 16:49:10--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-04-02/r2s.img.gz
Resolving ghproxy.com... 140.238.7.237
Connecting to ghproxy.com|140.238.7.237|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 109609993 (105M) [application/octet-stream]
Saving to: 'r2s.img.gz'

r2s.img.gz                                      100%[======================================================================================================>] 104.53M  11.1MB/s    in 9.9s    

2021-04-03 16:49:23 (10.6 MB/s) - 'r2s.img.gz' saved [109609993/109609993]

--2021-04-03 16:49:23--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-04-02/r2s.img.md5
[... 41 lines cut ...]
Saving config files...
备份文件已经写入，移除挂载
e2fsck 1.45.6 (20-Mar-2020)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
Padding at end of inode bitmap is not set. Fix? yes

rootfs: ***** FILE SYSTEM WAS MODIFIED *****
rootfs: 7706/32000 files (0.0% non-contiguous), 65115/128000 blocks
resize2fs 1.45.6 (20-Mar-2020)
Resizing the filesystem on /dev/loop0p2 to 3848192 (4k) blocks.
The filesystem on /dev/loop0p2 is now 3848192 (4k) blocks long.

正在打包...
开始写入，请勿中断...
FriendlyWrt.img (skip 0 bytes) -> /dev/mmcblk0 (O_DIRECT) (seek 0 bytes)
Source size is 15829303296 and dst size is 15829303296.
Copy 15829303296 bytes (14.74GiB)
block to read: 1048576, block total_read: 15829303296, skipped: 14849 blocks, written: 247 blocks
Worker 0: time: 54.6210, throughput: 276.38MiB/s, skipped: 14849 blocks, written: 247 blocks
total time: 54.6217, total throughput: 276.37MiB/s, total skipped: 14849 blocks, total written: 247 blocks
刷机完毕，正在重启...

DONGYUSEN, 2021-04-10:
估计是驱动问题？？