stupidloud/nanopi-openwrt #522: 部分网站解析不正常

dddfgeer, 2021-04-08:
就是有部分网站证书错误 解析不正常 就是一些国外冷门网站 麻烦看一下是什么原因

dddfgeer, 2021-04-08:
这些问题在ssrplus 还有hello world还有 passwall都有 可能是dns解析的问题

dddfgeer, 2021-04-08:
[image]

dddfgeer, 2021-04-08:
> 就是有部分网站证书错误 解析不正常 就是一些国外冷门网站 麻烦看一下是什么原因 r4s

klever1988, 2021-04-08:
开广告过滤https了吧

dddfgeer, 2021-04-08:
[image]

dddfgeer, 2021-04-08:
还没有启用这些功能 就是这些情况

dddfgeer, 2021-04-08:
显示证书错误

klever1988, 2021-04-08:
你开了什么插件不准备说一下吗

dddfgeer, 2021-04-08:
passwall ssrplus 还有hello world 全部都是这种情况 其他插件不开

dddfgeer, 2021-04-08:
这种问题是很早之前的版本就有这些问题了

dddfgeer, 2021-04-08:
应该是dns劫持的问题

klever1988, 2021-04-08:
我认为是你的机场节点的问题
不要用软路由，用exe去科学试一下

dddfgeer, 2021-04-08:
节点没有问题 换其他固件正常

dddfgeer, 2021-04-08:
这个bug难道没有办法了吗

nonozone, 2021-04-09:
顺便测试了你的两张截图的网址，第一个应该是cf节点有问题，第二个正常访问。

klever1988, 2021-04-09:
> 这个bug难道没有办法了吗

我把你这两个网址加到强制代理域名里，都可以正常访问，不能复现你的bug，要怎么解决

klever1988, 2021-04-09:
用slim版试一下

dddfgeer, 2021-04-09:
那个版本没有上网插件来的 passwall这些

dddfgeer, 2021-04-10:
不但是这些原因 还有部分国外冷门网站无法解析