stupidloud/nanopi-openwrt #524: 大佬R2S的20210402固件报错整理

lyjhyjyes, 2021-04-09:
hello world，helloworld开启，ssr+明明开关停止，却显示状态已启动。怎么会联动呢，虽然不影响使用，但是一开始让我以为ssr+没关
hello world和ssr+，明明设置jd和https://lv.ulikecam.com/和https://note.youdao.com/走直流，打开京东变成节点tw.jd.com，关闭代理开关后依然tw.jd.com。开代理后剪映无法使用识别字幕服务。网页有道云笔记无法打开。而之前minimal 11/27正常，翻墙代理设置我和以前一样没变化啊。
OpenClash，“Capsh异常，请尝试重新安装依赖【libcap】和相应的Capsh库，终止启动”，设置按照教程没有错。
ACC网络加速，修改设置后不一定能保存，总在刷新中。
网络共享ksmbd和以前一样的设置，挂载点正常，手机电脑无法发现共享点。旧版本能显示共享点。
Aria2各种设置，“已启用”，显示Aria2服务未运行，不知为何。
appfilter无法选择用户，select users没有选项。关闭acc，开启后appfilter，屏蔽无效。
感谢大佬的持续更新，辛苦了

klever1988, 2021-04-09:
关闭acc，开启后appfilter，屏蔽无效 。 这是appfilter和acc不兼容导致，appfiler作者已有说明
aria2故障原因，用新版应该能解决
ksmbd共享点未发现是因为wsdd2软件包未安装
oc需要安装libcap-bin
ssr和vssr很可能是因为服务进程名称相同所以判定为已启动
域名分流你留意下是不是浏览器cookie，以及网站判定区域时用的是哪些域名，或者采用其他代理模式

lyjhyjyes, 2021-04-09:
> 关闭acc，开启后appfilter，屏蔽无效 。 这是appfilter和acc不兼容导致，appfiler作者已有说明
> aria2故障原因，用新版应该能解决
> ksmbd共享点未发现是因为wsdd2软件包未安装
> oc需要安装libcap-bin
> ssr和vssr很可能是因为服务进程名称相同所以判定为已启动
> 域名分流你留意下是不是浏览器cookie，以及网站判定区域时用的是哪些域名，或者采用其他代理模式

谢谢大佬回复

lyjhyjyes, 2021-04-09:
> > 关闭acc，开启后appfilter，屏蔽无效 。 这是appfilter和acc不兼容导致，appfiler作者已有说明
> > aria2故障原因，用新版应该能解决
> > ksmbd共享点未发现是因为wsdd2软件包未安装
> > oc需要安装libcap-bin
> > ssr和vssr很可能是因为服务进程名称相同所以判定为已启动
> > 域名分流你留意下是不是浏览器cookie，以及网站判定区域时用的是哪些域名，或者采用其他代理模式
> 
> 谢谢大佬回复

使用passwall后，剪映和有道恢复正常。youtube打开流畅，只是mail.google.com打开超级缓慢，经常崩。而安卓上QQ邮箱app绑定谷歌邮箱，查看和回复邮件流畅。不知为何谷歌邮箱网页打不开

lyjhyjyes, 2021-04-09:
> 关闭acc，开启后appfilter，屏蔽无效 。 这是appfilter和acc不兼容导致，appfiler作者已有说明
> aria2故障原因，用新版应该能解决
> ksmbd共享点未发现是因为wsdd2软件包未安装
> oc需要安装libcap-bin
> ssr和vssr很可能是因为服务进程名称相同所以判定为已启动
> 域名分流你留意下是不是浏览器cookie，以及网站判定区域时用的是哪些域名，或者采用其他代理模式

使用passwall后，剪映和有道恢复正常。youtube打开流畅，只是mail.google.com打开超级缓慢，经常崩。而安卓上QQ邮箱app绑定谷歌邮箱，查看和回复邮件流畅。不知为何谷歌邮箱网页打不开

klever1988, 2021-04-09:
mail.google.com的页内引用了其他被q域名的css或者js，无法加载