stupidloud/nanopi-openwrt #526: 0409的ftp怎么没了？

1263478456, 2021-04-09:


1263478456, 2021-04-09:
并且在软件包安装的界面找不到 luci-app-vsftpd

klever1988, 2021-04-09:
好几天前拿掉了， 有需要用这个吗

1263478456, 2021-04-09:
有需要的，因为FTP传输速度要比SMB更快，并且由于我是PT站用户，平时在局域网内看一些比较大的电影的时候，使用SMB会造成卡顿。所以如果可以的话还请大佬加一下这个

siyang814, 2021-04-10:
我觉得大老加一个webdav吧， 比ftp香不是

LXY631, 2021-11-13:
> 有需要的，因为 FTP 传输速度要比 SMB 更快，并且由于我是 PT 站用户，平时在局域网内看一些比较大的电影的时候，使用 SMB 会造成卡顿。所以如果可以的话还请大佬加一下这个

请问下楼上，如何挂载使用ntfs格式的移动硬盘。
我使用2020-06-25 nanopi-r1s-h3的固件，可以自动挂载exFat格式的U盘，开启ftp正常使用，但是换ntfs格式的就是只读、无法写等操作，按照网上的教程 https://www.cnblogs.com/pcx69/p/3661271.html 设置挂载ntfs分区，安装 ntfs-3g插件，opkg install ntfs-3g，一直会报错
`
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ntfs-3g:
 *      kernel (= 5.10.78-1-27c01f3b5b6486f1446df2c7078e6fc0)
 * opkg_install_cmd: Cannot install package ntfs-3g.
`
不知道有没有好的方法可以解决，谢谢了。

klever1988, 2021-11-13:
> 请问下楼上，如何挂载使用ntfs格式的移动硬盘。 我使用2020-06-25 nanopi-r1s-h3的固件，可以自动挂载exFat格式的U盘，开启ftp正常使用，但是换ntfs格式的就是只读、无法写等操作，按照网上的教程 https://www.cnblogs.com/pcx69/p/3661271.html 设置挂载ntfs分区，安装 ntfs-3g插件，opkg install [...]

在op里对u盘格式化ntfs试试

LXY631, 2021-11-13:
> 在 op 里对 u 盘格式化 ntfs 试试

>> 在op里对u盘格式化。按照这个教程设置 https://iyzm.net/openwrt/72.html ，无奈没有 fdisk 命令，使用DiskGenius软件格式化的EXT4格式，插U盘测试正常可使用，谢谢了。