stupidloud/nanopi-openwrt #529: 大佬能不能让咱的R1也在线升级一下...

yydjh, 2021-04-09:


klever1988, 2021-04-09:
在终端执行
`cat /etc/board.json | jsonfilter -e '@["model"].id'`

然后复制一下内容过来

yydjh, 2021-04-09:
friendlyelec,nanopi-r1

yydjh, 2021-04-09:
谢谢大佬先..

yydjh, 2021-04-09:
md5sum.txt                          100%[=================================================================>]      45  --.-KB/s    in 0s      

2021-04-10 07:41:13 (1.75 MB/s) - 'md5sum.txt' saved [45/45]

今天固件已下载，准备解压
../truncate: line 1: syntax error: unterminated quoted string
到这里就这样了

klever1988, 2021-04-10:
`cat /sys/block/mmcblk0/size`
执行以下然后贴出结果，

另外r1的内存是512M吧？

yydjh, 2021-04-10:
15269888
我的这个是内存IG,8G存储

klever1988, 2021-04-10:
那你应该是脚本没有下载完全 先wget 再执行
wget https://raw.githubusercontent.com/klever1988/nanopi-openwrt/master/scripts/autoupdate.sh
sh autoupdate.sh

yydjh, 2021-04-10:
> 

>

yydjh, 2021-04-10:
[image]
[image]

yydjh, 2021-04-10:
[image]
[image]

klever1988, 2021-04-10:
你电报id发一下

yydjh, 2021-04-10:
大佬，我没有那个。。。。

klever1988, 2021-04-10:
qq

> 大佬，我没有那个。。。。

yydjh, 2021-04-10:
675084140

yydjh, 2021-04-21:
[image]
大佬，现在能到这一步了......

yydjh, 2021-04-21:
eMMC 的盘符是 /dev/mmcblk1