stupidloud/nanopi-openwrt #533: R2S固件运行一段时间会掉速

muronghan, 2021-04-10:
家里是电信500M宽带，刷完机后，运行正常，但在一段时间后，随着时间的推移和对界面的修改保存等，会逐渐掉速，比如测速测下来，从最初的550M，掉到470，再到360，再到280。重启后正常。

- 主路由或旁路由情况一样；
- 更换TF卡情况一样；
- 刷其他第三方R2S的固件，情况消失。

13311031900, 2021-04-10:
跟我R1S情况的一样,不知怎么回事.

klever1988, 2021-04-10:
可能是上行带宽被什么软件或者插件占了，不一定来自路由器

muronghan, 2021-04-10:
有意思的是，保存并应用后，速度会降下来，再保存并应用，再降，逐步降下来。目前试了三个第三方固件，除了这套，骷颅头的也有这个毛病，qiu simon的却没有，而后者是插件相对比较少的，比如没有VPN，网络存储菜单里只有一个USB打印服务器。骷髅头和qiu mison的内核很一致，但是初始插件多了太多，会不会是这个原因？如果能减少点插件，我可以试一下。

13311031900, 2021-04-10:
试过slim版本自己安装插件也有这个问题,不知道怎么回事

muronghan, 2021-04-10:
> 试过slim版本自己安装插件也有这个问题,不知道怎么回事

如果按你说法，就不大可能是插件过多所致，，，qiu版本是op21.02，难道是版本原因？。。我再试几个其他的第三个的固件看看

muronghan, 2021-04-11:
又做了一些测试，以下是记录：

- 本人是R2S，之前除了这套固件，还测了骷髅头和qiu simon的，测试方法是普通的设置，旁路由，插件打开了OpenClash，发现家宽电信500M下，本套固件会掉速到200M，骷髅头也差不多，而qiu simon（它是LAN-WAN互换的）不掉速。其间通过更换LAN-WAN口，换成主路由，更换交换机，换TF卡，甚至换电源和电源线，问题依旧。

- 因怀疑是固件版本低致，故又试了两个第三方固件，都是op21.02的，分别是第一个：https://github.com/soffchen/NanoPi-R2S，发现问题一样；第二个：https://github.com/RikudouPatrickstar/R2S-OpenWrt，这个是极简版，所以插件很少，发现也掉速，但是程度没那么严重，开启OpenClash后，掉到460M，LAN-WAN互换后，居然没掉速。

- 重新刷了本套4.11的最新固件r2s.img.gz，刷完后立即交换LAN-WAN口，不断地设置，保存并应用，先开了了Passwall，至此发现居然没掉速，重点来了，开启OpenClash后，一下子掉到200M以下。关闭OpenClash，速度恢复到500M以上。再换回LAN-WAN，速度不掉。

至此，先做个小结：
- R2S 原来的LAN口效率低，如果两口互换，作为旁路由速度更好；
- OpenClash开启后，会产生掉速，如果固件的插件少，那么掉速的程度低，qiu simon的开启了OpenClash之所以不会掉速，可能就是因为1）LAN-WAN互换；2）本身固件的插件就少；
- 关于掉速是相对的，我的是电信500M，如果是200M就无所谓掉速，如果是1Gbps，仍然可能是掉速，只不过我测不出来。

测速的软件是speedtest.net和电信自家的测速工具。

muronghan, 2021-04-11:
我目前关闭了OpenClash，开启了Hello World，测试一段时间看看。之前看到过视频，说OpenClash对CPU的占用比其他同类插件大，在CPU强的软路由里不明显，但是对CPU弱的软路由差别就很大。但是这个结论是基于代理的速度，并不是指和ISP直连的速度。

这个反映这个问题的，好像我是第一个？。。。我想可能有以下几个原因：
- 不是所有人都使用R1S/R2S/R4S；
- 不是所有人都使用本套固件（我自己用下来，对本固件是相对最满意的，bug比较少点，谢谢开发者的努力）；
- 不是所有人都有200M以上的家庭接入带宽；
- OpenClash设置起来相对麻烦，有不少替代产品；
- 不是所有人像我这样，天天关注着速度，其实200M和500M接入，在家庭使用上基本上是感觉不出差别的；
- 也许R4S的性能比较高，就没有我的问题，期待R4S的使用者也能做个测试。

xiaodinghoney, 2021-04-11:
我的R2S也是这样的问题。降速到90M

qazesz1, 2021-04-12:
去年12.14的最后一个用官方内核的固件会掉速吗

muronghan, 2021-04-13:
> 去年12.14的最后一个用官方内核的固件会掉速吗

不知道，，，你可以自己测啊

muronghan, 2021-04-13:
看到一年前的老帖子，好像有提到这个问题：https://github.com/vernesong/OpenClash/issues/300

现在不用tun模式，先用兼容模式，然后在实验性：绕过中国IP上打勾，先试一段时间看看。

qazesz1, 2021-04-13:
> > 去年12.14的最后一个用官方内核的固件会掉速吗
> 
> 不知道，，，你可以自己测啊

看样子是clash导致的问题，我一直用的ssrplus，而且网络带宽也没那么高，没有发现这个问题。只是iperf测过路由-pc发现改用immortal源码的系统跑不满940M，所以还留在稳定的12.14版本，等有机会再测下nat的性能如何

muronghan, 2021-04-17:
OpenClash 用兼容模式，然后在实验性：绕过中国IP上打勾，没有发现降速的情况，基本可以确定R2S上的降速是开启OpenClash引起，但如果OpenClash不去调用tun，并在实验性：绕过中国IP上打勾，就不产生降速现象。

closed 2021-04-17 by muronghan

michael-james0207, 2021-06-29:
> 又做了一些测试，以下是记录：
>
> * 本人是R2S，之前除了这套固件，还测了骷髅头和qiu simon的，测试方法是普通的设置，旁路由，插件打开了OpenClash，发现家宽电信500M下，本套固件会掉速到200M，骷髅头也差不多，而qiu simon（它是LAN-WAN互换的）不掉速。其间通过更换LAN-WAN口，换成主路由，更换交换机，换TF卡，甚至换电源和电源线，问题依旧。
> * [...]

我也是同样的问题，开启CLASH后掉速有时候会到100KB/S 人都要疯，而且最近还存在断网的情况，一直在研究这个，也和你一样测试了N+个固件，为此我还以为是硬件问题，买了一个树莓派回来测试，等产品到了我再回你

ladderhzy, 2022-01-20:
我是R4S,联通千兆，也出现这个问题，目前使用的方法只能是出现后，重启一下CLASH插件，然后继续用

xeasonwang, 2022-06-28:
> 我是R4S,联通千兆，也出现这个问题，目前使用的方法只能是出现后，重启一下CLASH插件，然后继续用

我是 R5S（用的不是这个固件），也有这个问题，而且不只是 OpenClash，Passwall、SSR+、HelloWorld 都有这个问题，只要软路由连续运行超过2天速度就会掉到只剩五分之一，4K 直接没法看，重启软路由又会恢复正常。同一节点 ClashX 测试又是正常的。
而且 Netdata 记录软路由 CPU 占用率全程不超过 40%，实在搞不懂。

esingtse, 2022-11-01:
> > 我是R4S,联通千兆，也出现这个问题，目前使用的方法只能是出现后，重启一下CLASH插件，然后继续用
> 
> 我是 R5S（用的不是这个固件），也有这个问题，而且不只是 OpenClash，Passwall、SSR+、HelloWorld 都有这个问题，只要软路由连续运行超过2天速度就会掉到只剩五分之一，4K 直接没法看，重启软路由又会恢复正常。同一节点 ClashX 测试又是正常的。 而且 Netdata 记录软路由 CPU 占用率全程不超过 40%，实在搞不懂。

R2S同样，openclash都关了，一两天就从300M掉到100M，只能通过重启

huashiren88, 2025-03-05:
我是N5105，2000M宽带，也换了很多个固件，困扰我3年了，一直都有这个问题，反正开机是必跑满速，用一段时间，测速就上不去了，一直找不到原因。只开了openclash其他都没开。CPU和内存占用都很小。

esingtse, 2025-03-06:
不知道是不是R2S的缺陷还是什么,无论当主路由或者旁路由都不太稳定,现在都放柜子里面吃灰了

esingtse, 2025-03-06:
现在tplink当主路由拨号,然后x86虚拟机搞个istore OS当旁路由,竟然出奇地稳定,主路由不掉速,不断网,有需要科学的设备再改一下网关到旁路由,很香