stupidloud/nanopi-openwrt #534: 为啥我SSH在线更新最近的固件时就会失败？

magihub, 2021-04-10:
我SSH在线更新最近的固件时就会失败，然后只能重新卡刷回3/28的固件。
[image]

而在我卡刷3/28固件，或者最新的固件的话，也无法手动用diskgenius扩展分区，提示错误。（我线刷，也主要是为了扩展分区大小）
[image]

magihub, 2021-04-10:
而且，我如果直接卡刷最近两天的固件，使用我4/1备份的配置文件，又会造成故障。而我同样的配置文件，在3/28固件上就没有问题。

klever1988, 2021-04-10:
卡刷之后可以试一下在linux(openwrt也可以)下对分区进行扩容，把刷好固件的卡用读卡器接入，软件使用cfdisk e2fsck resize2fs，在网上可以找到相应教程

使用备份进行还原的话，可能是备份的内容和新版的插件产生某些冲突了，可以考虑解压编辑备份的tar.gz文件，只保留需要备份的插件的配置，删除其它，然后重新打包再还原

至于在线升级的故障，我这边r1s r2s都没有问题，无法复现故障。你第一个截图可以再网上拉一点，看看下载解压的过程是不是正常完成了