stupidloud/nanopi-openwrt #546: docker 能用吗运行报错

engine-we, 2021-04-11:
docker: Error response from daemon: OCI runtime create failed: container_linux.go:367: starting container process caused: process_linux.go:352: getting the final child's pid from pipe caused: EOF: unknown.
ERRO[0001] error waiting for container: context canceled

这是啥问题 我是小白

alezh, 2021-04-15:
我R4S也是这个问题。

eagleonefight, 2021-04-15:
我也是这个问题

onlypeng, 2021-04-16:
R2S 同上 不能用

zhixiangjoy, 2021-04-17:
我也用不了，希望大佬赶紧修复啊
Containers: start jd_scripts...code:500 OCI runtime create failed: container_linux.go:367: starting container process caused: process_linux.go:352: getting the final child's pid from pipe caused: EOF: unknown

klever1988, 2021-04-18:
我也不知道到底缺了啥配置  会的人请来个pr吧

UlysessWeb, 2021-04-19:
一摸一样的问题，查了半天也没解决

lissettecarlr, 2021-04-30:
额，原来大家都有这个问题，我也是捣鼓好半天，找不到docker服务咋启动，etc/init.d里面没有docker