stupidloud/nanopi-openwrt #547: 局域网内使用动态DNS域名无法访问对应内网IP端口

liruishengus, 2021-04-12:
我是用oray的动态DNS域名，在0409版本之前都是正常可以在内网使用域名来访问对应的NAS，可以通过WiFi来同步群晖的moment...
但是升级到0410版本之后，就不可以了，R2S的动态DNS插件更新的IP是正常的，ping动态DNS域名是反馈正确的IP地址，但就是无法通过内网访问动态域名5000端口（我已经在防火墙里面设置了5000端口转发到内网群晖），现在每次都是要手动改回群晖的内网IP才可以同步...
总觉得是不是0410版本之后有个设置限制了内网访问R2S外网IP端口回不来呢？

jennergray, 2021-04-12:
我也是一样的问题，这样在内网访问nas太麻烦了。只能在dnsmasq强制更新一条规则映射内网的域名和ip，大佬有没有办法能解决啊。

klever1988, 2021-04-12:
docker引发的问题 我另起一个带docker的版本 默认不带

jennergray, 2021-04-12:
> docker引发的问题 我另起一个带docker的版本 默认不带

你好，现在使用的是不带docker的版本，是不带docker的版本会有这个问题吗？意思是用带docker的版本能解决这个吗？

biliwala, 2021-04-13:
https://github.com/coolsnowwolf/lede/issues/2103#issuecomment-751424989

klever1988, 2021-04-13:
> [coolsnowwolf/lede#2103 (comment)](https://github.com/coolsnowwolf/lede/issues/2103#issuecomment-751424989)

感谢！

liruishengus, 2021-04-14:
更新到0413版本之后恢复正常了，局域网内可以通过DDNS域名访问端口了

klever1988, 2021-04-14:
> 更新到0413版本之后恢复正常了，局域网内可以通过DDNS域名访问端口了

你使用的是docker版固件还是不含docker?

jennergray, 2021-04-14:
> > 更新到0413版本之后恢复正常了，局域网内可以通过DDNS域名访问端口了
> 
> 你使用的是docker版固件还是不含docker?

我这边用不带docker的13号的版本还是不行的

liruishengus, 2021-04-17:
> > 更新到0413版本之后恢复正常了，局域网内可以通过DDNS域名访问端口了
> 
> 你使用的是docker版固件还是不含docker?

目前我是用不含docker版本0413版本，局域网内恢复域名访问了