stupidloud/nanopi-openwrt #554: 新版本r4s固件无法正常启动passwall还有ssrplus hello world正常

dddfgeer, 2021-04-14:


klever1988, 2021-04-14:
看一下插件日志有什么输出

dddfgeer, 2021-04-14:
2021-04-15 01:52:53: 删除相关防火墙规则完成。
2021-04-15 01:52:56: 重启 dnsmasq 服务[0]
2021-04-15 01:52:56: 清空并关闭相关程序和缓存完成。
2021-04-15 01:52:57: TCP节点：🇭🇰 Hong Kong 10，节点：untmkksxp.vensas.info:5605，监听端口：1041
2021-04-15 01:52:57: UDP节点：🇭🇰 Hong Kong 10，节点：untmkksxp.vensas.info:5605，监听端口：1051
2021-04-15 01:52:57: 过滤服务配置：准备接管域名解析...
2021-04-15 01:52:57:   + [0]Pdnsd (127.0.0.1:7913)...
2021-04-15 01:52:57:   | - [0]上游DNS：8.8.8.8:53
2021-04-15 01:52:57:   - 域名解析：pdnsd + 使用(TCP节点)解析域名...
2021-04-15 01:52:57:   * 请确认上游 DNS 支持 TCP 查询，如非直连地址，确保 TCP 代理打开，并且已经正确转发！
2021-04-15 01:52:57:   - [0]节点列表中的域名(vpsiplist)：202.96.128.86,202.96.134.133
2021-04-15 01:52:57:   - [0]域名白名单(whitelist)：202.96.128.86,202.96.134.133
2021-04-15 01:52:57:   - [0]节点订阅域名(whitelist)：202.96.128.86,202.96.134.133
2021-04-15 01:52:57:   - [0]代理域名表(blacklist)：127.0.0.1#7913
2021-04-15 01:52:57:   - [0]防火墙域名表(gfwlist)：127.0.0.1#7913
2021-04-15 01:52:59:   - [0]中国域名表(chnroute)：202.96.128.86,202.96.134.133
2021-04-15 01:52:59: 开始加载防火墙规则...
2021-04-15 01:52:59: 加入负载均衡的节点到ipset[vpsiplist]直连完成
2021-04-15 01:52:59: 加入所有节点到ipset[vpsiplist]直连完成
2021-04-15 01:52:59: 加载路由器自身 TCP 代理...
2021-04-15 01:52:59:   - [1]将上游 DNS 服务器 8.8.8.8:53 加入到路由器自身代理的 TCP 转发链2
2021-04-15 01:52:59: 加载路由器自身 UDP 代理...
2021-04-15 01:52:59:   - [0]不代理 UDP 端口：80,443
2021-04-15 01:52:59: TCP默认代理：使用TCP节点 [中国列表以外](REDIRECT:1041)代理所有端口
2021-04-15 01:53:00: UDP默认代理：使用UDP节点 [中国列表以外](TPROXY:1051)代理除80,443外的所有端口
2021-04-15 01:53:00: 防火墙规则加载完成！
2021-04-15 01:53:02: 重启 dnsmasq 服务[0]
2021-04-15 01:53:02: 运行完成！

dddfgeer, 2021-04-14:
[image]

klever1988, 2021-04-14:
看不出来什么问题  

ssrp什么情况

dddfgeer, 2021-04-15:
也是一样 无法连接

dddfgeer, 2021-04-15:
还有smart dns我不会设置 折腾了一个晚上

closed 2021-04-15 by klever1988