stupidloud/nanopi-openwrt #559: 本固件使用大半年，常常出现定时重启失败。

AlreadyM, 2021-04-16:
该R2S设备扮演角色为光猫桥接下的一级路由，，网络时常卡滞。定时重启任务也是为了解决卡滞定制的。
疑点：
①重启失败是固件的缺陷还是硬件的缺陷
②硬件的缺陷可能出在什么地方，如网卡的掉线、设备计算性能突然拉闸（就像windows突然卡死）
③固件缺陷可能是出在什么地方，固件对硬件的联系丢失、固件内功能逻辑冲突

确定点：网络卡滞不是定时重启任务失败导致的，之前就存在该卡滞现象。
要解决本问题，我知道是非常困难的。因为这家伙不是随时可复现，偶尔3天偶尔一周。但还请开发大佬常记得有个 影响长期静默运行的问题存在。

supercaizehua, 2021-04-16:
r1s-h5 r2s 会有断流的情况发生
表现形式为
第一种，192.168.2.1 打不开，确认下级AP功能正常运行，感觉像是机子死机了一样
第二种，192.168.2.1 能打开，网络接口那边WAN口需要手动重新连接一下又能继续正常上网。
以上两种情况是在春节回来以后更新了固件发生的（非在线升级），不知道哪个版本固件开始的，老版本固件没什么问题
今天准备更新一下最新固件，如果问题解决了会过来反馈

zanderzhng, 2021-04-17:
> r1s-h5 r2s 会有断流的情况发生
> 表现形式为
> 第一种，192.168.2.1 打不开，确认下级AP功能正常运行，感觉像是机子死机了一样
> 第二种，192.168.2.1 能打开，网络接口那边WAN口需要手动重新连接一下又能继续正常上网。
> 以上两种情况是在春节回来以后更新了固件发生的（非在线升级），不知道哪个版本固件开始的，老版本固件没什么问题
> 今天准备更新一下最新固件，如果问题解决了会过来反馈

我有遇到类似的情况。
断网后短时间内通常2.1能进去，进去后能观察到2.1断开，但是不及时给它重启的话过会儿2.1就也进不去了。
我也在尝试更新固件来解决这个问题，但是因为发生的不是特别频繁（一周一次的样子），所以不太好追踪。

zhutodd, 2021-07-26:
手动断电重启后，插件经常性不行自动运行起来，比如ssr,还有广告屏蔽等