stupidloud/nanopi-openwrt #566: Cannot log in via Luci interface, SSH only. NanoPi R4S

dallascaudell, 2021-04-18:
I unplugged the router for a couple of days, last successful login was 4/15/21. Now when I try to login the interface just resets with no error. If I enter the wrong password it says wrong username or password, but when I enter the correct it just resets and stays on the login screen. I am able to login via SSH. Any way I can fix this? I left my SD card reader at work :(

klever1988, 2021-04-19:
you can check the logs , try to see what's going on when you login from Luci.

dallascaudell, 2021-04-19:
> you can check the logs , try to see what's going on when you login from Luci.

I figured it out. I couldn't log in using https://, I had to use http://. I reinstalled fresh and now I can log in via https:// but TTYD only works when viewing via http://. Is that normal?

klever1988, 2021-04-19:
you should contact the app author I think

dallascaudell, 2021-04-19:
Will do