stupidloud/nanopi-openwrt #567: 大佬你好，网件A6210 USB无线网卡的驱动能加入R2S吗？芯片型号貌似是MT7612U

yangwen9, 2021-04-19:
大佬你好，网件A6210 USB无线网卡的驱动能加入R2S去吗吗？芯片型号貌似是MT7612U，谢谢了！目前的固件对华硕USB-AC57的无线网卡驱动很完美，如果能支持网件的无线网卡那更好了

klever1988, 2021-04-19:
加了 试试

yangwen9, 2021-04-20:
好的，感谢大佬，我晚上回家下个最新的R2S固件版本试试

yangwen9, 2021-04-20:
4月20号刚release的R2S（r2s.img.gz）固件怎么用balenaEtcher刷的时候，大多时候刷写失败，偶尔能刷写成功一次。 4月20日以前的固件我随机试了几个，没有刷failed问题。

yangwen9, 2021-04-20:
感谢klever 大佬，网件A6210 无线网卡能正常驱动了，信号也不错，现在唯一的小问题就是无线网卡的指示灯没能点亮，不过不影响使用，看您能不能把这个bug通过优化fix

jielis, 2021-05-21:
> 感谢klever大佬，网件A6210无线网卡能正常驱动了，信号也不错，现在唯一的小问题就是无线网卡的提示没能点亮，不过不影响使用，看您能不能把这个bug通过优化修复

您好，麻烦您能不能教教我如何设置无线啊？我买了一张大佬推荐的COMFAST 811AC网卡，但是不会设置，设置上密码之后一直用手机连接一直不能连接成功，然后一通设置之后，现在无线都打不开了，直接显示无线未开启或未关联

yangwen9, 2021-05-21:
模式AC、信道149、频宽80， 参考我下面设定试试，记得保存&应用时多点几次，有时候点一次保存不上，保存&应用后，软重启下，然后再看看，不行就刷个新的固件了，有时候跟你SD卡也有关系，R2S很挑SD卡，很多莫名其妙的问题都跟SD卡有关系
[image]

Ro0kie, 2023-11-18:
> 模式AC、信道149、频宽80， 参考我下面设定试试，记得保存&应用时多点几次，有时候点一次保存不上，保存&应用后，软重启下，然后再看看，不行就刷个新的固件了，有时候跟你SD卡也有关系，R2S很挑SD卡，很多莫名其妙的问题都跟SD卡有关系 [image]

你好，请问你的网件A6210 也是这样设置的吗？？ 另外有推荐的sd卡品牌和型号吗，谢谢~